coolsnowwolf/lede #1125: 3月21日更新的代码 mtk7621 编译失败。

wonathan, 2019-03-21:
scripts/Makefile.build:326: recipe for target '/home/wonathan/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2019-03-16-22364904/tx.o' failed
make[5]: *** [/home/wonathan/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2019-03-16-22364904/tx.o] Error 1
Makefile:1533: recipe for target '_module_/home/wonathan/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2019-03-16-22364904' failed
make[4]: *** [_module_/home/wonathan/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2019-03-16-22364904] Error 2
make[4]: Leaving directory '/home/wonathan/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.106'
Makefile:236: recipe for target '/home/wonathan/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2019-03-16-22364904/.built' failed
make[3]: *** [/home/wonathan/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt76-2019-03-16-22364904/.built] Error 2
make[3]: Leaving directory '/home/wonathan/lede/package/kernel/mt76'
time: package/kernel/mt76/compile#4.28#1.14#5.70
package/Makefile:107: recipe for target 'package/kernel/mt76/compile' failed
make[2]: *** [package/kernel/mt76/compile] Error 2
make[2]: Leaving directory '/home/wonathan/lede'
package/Makefile:103: recipe for target '/home/wonathan/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/wonathan/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/wonathan/lede'
/home/wonathan/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

chd1998, 2019-03-22:
D2遇到同样错误。。。

wonathan, 2019-03-22:
> D2遇到同样错误？

同是d2

RC1844, 2019-03-22:
https://github.com/coolsnowwolf/lede/issues/1127#issuecomment-475466866
手动更新Makefile即可

chd1998, 2019-03-22:
> [#1127 (comment)](https://github.com/coolsnowwolf/lede/issues/1127#issuecomment-475466866)
> 手动更新Makefile即可

可有详细的说明？谢谢！

RC1844, 2019-03-22:
直接修改package/kernel/mt76/Makefile更新部分，重新编译

chd1998, 2019-03-22:
OK

chd1998, 2019-03-22:
改为这个：
PKG_SOURCE_DATE:=2019-03-03
PKG_SOURCE_VERSION:=461f3b0930b0fd9972a434289d7420519634294d
PKG_MIRROR_HASH:=9d9e8775ee0f0d2a48de9d7b05dfeb50d018126fc1557600a296b9ec2cdaf7ad

hcym, 2019-03-22:
一模一样，mt76没搞好，用官方的没事

closed 2019-03-22 by coolsnowwolf

yunmi323, 2019-03-22:
> 直接修改package/kernel/mt76/Makefile更新部分，重新编译

更新了还是一样的问题怎么办