coolsnowwolf/lede #1127: K2  编译mt76-2019-03-16-22364904/tx.c时出现缺少参数错误

RC1844, 2019-03-21:
CC [M]  /home/rcs/lede1/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/mt76-2019-03-16-22364904/mt76x02_mmio.o
/home/rcs/lede1/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/mt76-2019-03-16-22364904/tx.c: In function 'mt76_txq_schedule_list':
/home/rcs/lede1/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/mt76-2019-03-16-22364904/tx.c:528:4: error: too few arguments to function 'ieee80211_return_txq'
    ieee80211_return_txq(dev->hw, txq);
    ^~~~~~~~~~~~~~~~~~~~
In file included from /home/rcs/lede1/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/mt76-2019-03-16-22364904/mt76.h:27:0,
                 from /home/rcs/lede1/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/mt76-2019-03-16-22364904/tx.c:17:
/home/rcs/lede1/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211/net/mac80211.h:6127:1: note: declared here
 ieee80211_return_txq(struct ieee80211_hw *hw, struct ieee80211_txq *txq,
 ^~~~~~~~~~~~~~~~~~~~
  CC [M]  /home/rcs/lede1/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/mt76-2019-03-16-22364904/mt76x02_txrx.o
make[5]: *** [scripts/Makefile.build:326: /home/rcs/lede1/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/mt76-2019-03-16-22364904/tx.o] Error [1](url)
make[5]: *** Waiting for unfinished jobs....
make[4]: *** [Makefile:1533: _module_/home/rcs/lede1/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/mt76-2019-03-16-22364904] Error `2`

dongjuanyong, 2019-03-22:
https://github.com/openwrt/openwrt/commit/ca9ad880f29f3ef9eee9bb0342610d6f879e7770

> PKG_SOURCE_DATE:=2019-03-19
> PKG_SOURCE_VERSION:=e7bd802cf728f76ce42b6b7529db14e7fc027f0c
> PKG_MIRROR_HASH:=93ad26c322fe06e57c321b9d6f4332a2a2d626a4adcfe316465216b5216fbec6

RC1844, 2019-03-22:
> [openwrt / openwrt @ ca9ad88](https://github.com/openwrt/openwrt/commit/ca9ad880f29f3ef9eee9bb0342610d6f879e7770)
> 
> > PKG_SOURCE_DATE：= 2019-03-19 
> > PKG_SOURCE_VERSION：= e7bd802cf728f76ce42b6b7529db14e7fc027f0c 
> > PKG_MIRROR_HASH：= 93ad26c322fe06e57c321b9d6f4332a2a2d626a4adcfe316465216b5216fbec6

谢谢，其实昨晚已尝试自行添加参数，编译通过了，但看你提供的发现我添加了相反的参数。。。。

closed 2019-03-22 by RC1844