coolsnowwolf/lede #1134: v2ray 服务器配置错误

ytzong, 2019-03-26:
ssr plus+ 里服务器参数配置如下：

加密方式：NONE
Transport：TCP
Camouflage Type：HTTP
HTTP Host：www.bing.com
                      cn.bing.com
HTTP Path：/

通过 top 命令看到配置文件是这个 /var/etc/v2-ssr-retcp.json，查看里面的内容有一段是这样的
```
     "streamSettings": {
       "network": "tcp",
       "tlsSettings": {
         "allowInsecure": false
       },
       "security": "none"
     }
```
HTTP 配置的那些参数都丢了，正确的应该类似于这样
```
"streamSettings": {
	"network": "tcp",
	"tcpSettings": {
		"header": {
			"type": "http",
			"request": {
				"version": "1.1",
				"method": "GET",
				"path": ["/"],
				"headers": {
					"Host": ["www.bing.com", "cn.bing.com"],
					"User-Agent": [
						"Mozilla/5.0 (Windows NT 10.0; WOW64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/55.0.2883.75 Safari/537.36",
						"Mozilla/5.0 (iPhone; CPU iPhone OS 10_0_2 like Mac OS X) AppleWebKit/601.1 (KHTML, like Gecko) CriOS/53.0.2785.109 Mobile/14A456 Safari/601.1.46"
					],
					"Accept-Encoding": ["gzip, deflate"],
					"Connection": ["keep-alive"],
					"Pragma": "no-cache"
				}
			}
		}
	}
}
```
希望最好能开放自定义 json 的功能，多谢

iotsys, 2019-03-27:
最终生成的配置文件不是 "/tmp/config.json" ？我一直以为是这个

ytzong, 2019-03-27:
> 最终生成的配置文件不是 "/tmp/config.json" ？我一直以为是这个（走错片场了，我说的 v2ray）
top 命令看进程运行的是
```
/usr/bin/v2ray/v2ray -config /var/etc/v2-ssr-retcp.json
```
我说的是 ssr plus+ 三合一

505575853, 2019-03-27:
[ssr plus.zip](https://github.com/coolsnowwolf/lede/files/3013797/ssr.plus.zip)

试试这个

ytzong, 2019-03-27:
@505575853 安装的时候提示分别缺少 luci-lib-fs、kcptun、luci-app-filetransfer，其中 kcptun 我在软件包里安装了 luci-app-kcptun，luci-app-filetransfer 应该就是压缩包里的文件，不过它缺少 luci-lib-fs，我搜了一下没找到下载的地方，请问这个到哪里下载？

505575853, 2019-03-27:
[luci-lib-fs_1.0-1_all.zip](https://github.com/coolsnowwolf/lede/files/3015644/luci-lib-fs_1.0-1_all.zip)

ytzong, 2019-03-28:
@505575853 谢大佬，还差  kcptun， luci-app-kcptun 不行，请问这些文件是怎么找的啊

sbily1988, 2019-04-03:
我也有这个问题，不管是v2Ray pro  还是SSR plus，如果使用TCP-HTTP混淆就有问题，v2Ray pro表现的是v2Ray core直接无法启动，ssr- plus表现的是 v2Ray core启动了，但是无法正常翻墙，希望v2Ray pro或者SSR plus可以增加支持VMESS链接添加节点

crotoc, 2019-04-03:
> 我也有这个问题，不管是v2Ray pro 还是SSR plus，如果使用TCP-HTTP混淆就有问题，v2Ray pro表现的是v2Ray core直接无法启动，ssr- plus表现的是 v2Ray core启动了，但是无法正常翻墙，还希望能早日解决，另外希望v2Ray pro或者SSR plus可以增加支持VMESS链接添加节点

估计开发bandwith不够，懂bash的话可以自己随便改改。你知道vess链接的编写方式么？

coolsnowwolf, 2019-04-18:
确实没做这个功能，希望用到的同学，加到 genv2config.lua 这个里面

coolsnowwolf, 2019-04-18:
另外，v2ray pro 属于放弃了的项目，不再更新，只会更新三个一 ssr plus

skyunivs, 2019-08-21:
@ytzong 根据你的配置，问题出在"tcpSettings"这里， 可以在源文件中找到文件“～/lede/package/lean/luci-app-ssr-plus/root/usr/share/shadowsocksr/genv2config.lua”
按类似语法加上"tcpSettings"这一项， 但还不能按你的配置文件细化来加，经试验加到type:http这儿即可. 后编译试了下是可以的。进一步加具体点行不行不知道。
s s r p l u s 中通过填参数方式加节点还是比较方便的（方向是好的）， 但因v2配置较复杂， 容易出错的。

closed 2019-09-29 by coolsnowwolf