coolsnowwolf/lede #1151: 没有人觉得这里代码编译出来的，进管理界面各种不顺吗？

davidtwgit, 2019-04-09:
安装op官方的系统还好，
但是为了大家都知道的那啥，想用这个

几个路由器编译了很多次，进web管理界面很慢，常常bad gateway

希望出基于官方代码的lean 包，非常感谢

redchenjs, 2019-04-09:
建议自行移植

davidtwgit, 2019-04-09:
感觉主要问题还是在于不断的升级版本，都是无关痛痒的升，然而各种升级的后果就是对cpu专用越来越大。。。有没有人告诉我这里酸酸乳相对完美，没有明显毛病的最低版本，checkout到哪个点？谢谢

davidtwgit, 2019-04-09:
embedded  devices不是电脑，动不动bump。。。性能还比不上人家2.x，3.x内核的原厂固件。。。ok？！：”“：！？：”

redchenjs, 2019-04-09:
。。。提到酸酸乳那还跟原厂固件比个什么劲，根本就是两种需求，用第三方固件还不是一个玩得开心就好，本来就没售后服务。

davidtwgit, 2019-04-09:
酸酸乳相对完美，没有明显毛病的最低版本，checkout到哪个点呢？

davidtwgit, 2019-04-09:
哎，没法玩的开心啊，设置界面太卡，是不是该琢磨下不要设置后台，全部写config文件到ssh来设置

uglee, 2019-04-09:
自己去取lean那个packages ，然后直接放到自己的低版本里面去make就行了。lede应该是没啥问题的。再低没试过不清楚

qyh214, 2019-04-09:
这个issue不太完整，不提供基于哪个commit，也没说明是在编译哪个路由器固件时遇到的问题。

在非常早期，2017年某段时间里的确出现了性能低下，但之后就非常完美了。

我自用的网件r7800，给基友用的华硕ac58u、acrh17都非常完美，使用很舒服。

另外，issue不是用于版聊和抱怨的，请提供具有建设性的问题。

lan542662, 2019-04-10:
1024信仰版

closed 2019-04-15 by coolsnowwolf