coolsnowwolf/lede #1152: 出现GOlang 错误怎么办？

dukmonki, 2019-04-09:


hjw210, 2019-04-10:
( cd /home/parallels/lede/build_dir/hostpkg/go-1.12.1/src ; GOROOT_FINAL=/home/parallels/lede/staging_dir/hostpkg/lib/go-cross  GOROOT_BOOTSTRAP=/home/parallels/lede/build_dir/hostpkg/go-1.12.1/.go_bootstrap CC=gcc CXX=g++  bash make.bash --no-banner ; )
Building Go cmd/dist using /home/parallels/lede/build_dir/hostpkg/go-1.12.1/.go_bootstrap.
Building Go toolchain1 using /home/parallels/lede/build_dir/hostpkg/go-1.12.1/.go_bootstrap.
Building Go bootstrap cmd/go (go_bootstrap) using Go toolchain1.
Building Go toolchain2 using go_bootstrap and Go toolchain1.
Building Go toolchain3 using go_bootstrap and Go toolchain2.
go build cmd/compile/internal/ssa: /home/parallels/lede/build_dir/hostpkg/go-1.12.1/pkg/tool/linux_amd64/compile: signal: killed
go tool dist: FAILED: /home/parallels/lede/build_dir/hostpkg/go-1.12.1/pkg/tool/linux_amd64/go_bootstrap install -gcflags=all= -ldflags=all= -a -i cmd/asm cmd/cgo cmd/compile cmd/link: exit status 1
Makefile:271: recipe for target '/home/parallels/lede/build_dir/hostpkg/go-1.12.1/.built' failed
make[3]: *** [/home/parallels/lede/build_dir/hostpkg/go-1.12.1/.built] Error 2
make[3]: Leaving directory '/home/parallels/lede/feeds/packages/lang/golang/golang'
time: package/feeds/packages/golang/host-compile#157.15#31.60#314.62
package/Makefile:107: recipe for target 'package/feeds/packages/golang/host/compile' failed
make[2]: *** [package/feeds/packages/golang/host/compile] Error 2
make[2]: Leaving directory '/home/parallels/lede'
package/Makefile:103: recipe for target '/home/parallels/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/parallels/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/parallels/lede'
/home/parallels/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

我出现这个问题

dukmonki, 2019-04-10:
不知道怎么解决

zxlhhyccc, 2019-04-10:
你是用什么编译的？musl、ublic还是glibc？musl应该没问题，但是后面的就有问题。。。不知道怎么解决！

zxlhhyccc, 2019-04-10:
我科学上网也是不行，好像无法下载相应的文件。。。

zxlhhyccc, 2019-04-13:
我用VPS编译过去了，是依赖包被墙了，必须国外的IP可以下载！

closed 2019-04-15 by coolsnowwolf

zfdang, 2020-04-27:
应该是内存太小， go的工具被系统杀掉了。换个大内存的地方就成了