coolsnowwolf/lede #1153: 最新源码编译的斐讯K3固件，V2ray不能运行

yasum2006, 2019-04-09:
L大，最新源码编译的斐讯K3固件，V2ray不能运行，日志提示：2019-04-09 21:57:37  Started!
而编译的WRT1900AC  V2ray运行正常，非常喜欢你的openwrt固件，谢谢你为网友带来的很多乐趣

shenzt68, 2019-04-10:
同样问题

cnsilvan, 2019-04-11:
估计是 v2ctl的缺失

cnsilvan, 2019-04-11:
下载[https://github.com/v2ray/v2ray-core/releases/download/v4.18.0/v2ray-linux-arm.zip](url)
把v2ray 和 v2ctl 放入/usr/bin/v2ray/
chmod +x 它们
重启服务即可

coolsnowwolf, 2019-04-12:
K3 的处理器不带 FPU，所以无法运行 ARMv7 的 bin （ Golang 需要 FPU）

但 OpenWrt 将这个 CPU 判断为 ARM v7 导致的

LGA1150, 2019-04-12:
用这个 makefile 从源码编译 golang 软件包
https://github.com/kuoruan/openwrt-v2ray
已测试 MIPS 不需要内核 FPU 模拟也可以运行

shenzt68, 2019-04-15:
> 用这个 makefile 从源码编译 golang 软件包
> https://github.com/kuoruan/openwrt-v2ray
> 已测试 MIPS 不需要内核 FPU 模拟也可以运行

你确认k3也能用吗？
Cleaning module download cache (golang/go#27455)
go: finding v2ray.com/core/.go_work/build latest
go: finding v2ray.com/core/.go_work latest
can't load package: package v2ray.com/core/.go_work/build: unknown import path "v2ray.com/core/.go_work/build": cannot find module providing package v2ray.com/core/.go_work/build

LGA1150, 2019-04-15:
@shenzt68
请使用官方源码编译，lean 的 golang 好久没更新了

shenzt68, 2019-04-15:
@LGA1150
就是用官方编译的，通不过啊！

LGA1150, 2019-04-15:
要挂梯子？

shenzt68, 2019-04-15:
@LGA1150 就在墙外

SouHong, 2019-04-16:
@shenzt68我遇到类似情况过，有时重新编译就可以正常通过了。

hcym, 2019-04-16:
n1盒子也是，上礼拜还行的，昨天编译了就挂了

zxlhhyccc, 2019-04-17:
@shenzt68，要挂梯子，而且必须是在路由里设置SSR全局。对了，我编译了你的k3screenctrl的源码后，屏幕中的版本和MAC、温度的那个屏幕中MAC显示不了，而且固件更新的那个屏幕屏幕新版显示的是MAC地址而不是版本，其他均显示正常！

shenzt68, 2019-04-17:
@zxlhhyccc  我的k3screenctrl源码只能这样了，我也查不出原因，基本能用就行了。另外，你编译过lean的v2ray插件（用于K3的），我怎么都不行。编译机器在墙外，与翻墙无关。

zxlhhyccc, 2019-04-17:
@shenzt68 你的k3screenctrl源码不显示MAC，我查看了一下能显示MAC的并已编译好了的usr/bin/k3screenctrl,发现里面除了HW VERSION、FW VERSION、MAC ADDRESS外，还有一个NEW VERSION
[image]

barton0426, 2019-04-22:
> K3 的处理器不带 FPU，所以无法运行 ARMv7 的 bin （ Golang 需要 FPU）
> 
> 但 OpenWrt 将这个 CPU 判断为 ARM v7 导致的

那么自行编译一个可用的v6版本然后替换可以吗

coolsnowwolf, 2019-04-23:
是可以的

zxlhhyccc, 2019-04-30:
> > K3 的处理器不带 FPU，所以无法运行 ARMv7 的 bin （ Golang 需要 FPU）
> > 但 OpenWrt 将这个 CPU 判断为 ARM v7 导致的
> 
> 那么自行编译一个可用的v6版本然后替换可以吗

用于v6的版本哪里有下载？

zxlhhyccc, 2019-05-01:
> 下载[https://github.com/v2ray/v2ray-core/releases/download/v4.18.0/v2ray-linux-arm.zip](url)
> 把v2ray 和 v2ctl 放入/usr/bin/v2ray/
> chmod +x 它们
> 重启服务即可
我用大雕以前的源码编译，和你这个其实是一样，不能运行。。。。

closed 2019-05-02 by coolsnowwolf

GOUKI9999, 2019-06-13:
哪位有编译好的v6版本请赐下一份……