coolsnowwolf/lede #1170: v2ray和frpc太大，目前有大佬指教了压缩v2ray，但frpc不知道怎么压缩，目前无法编译进16M闪存，如何压缩？

zxlhhyccc, 2019-04-14:


dongjuanyong, 2019-04-14:
这个，没想象的这么难，举一反三挺容易的
https://gist.github.com/dongjuanyong/88eadaa3c94fa54fa56b66f188d46cab
试了试：
```
        File size         Ratio      Format      Name
   --------------------   ------   -----------   -----------
  10393440 ->   2892252   27.83%   linux/amd64   frpc

```
go语言这二进制文件的效率真的堪忧，kcptun也是一样

LGA1150, 2019-04-14:
折中压缩真的有效？在生成 squashfs 时已经经过了 XZ 最大压缩

dongjuanyong, 2019-04-14:
> 这种压缩真的有效？在生成 squashfs 时已经经过了 XZ 最大压缩

说实话我不知道，因为我是x86_64的软路由，用的ext4+uefi。而且root分区随便设，我也用不着压缩

zxlhhyccc, 2019-04-14:
谢谢dongjuanyong大佬，搞定了，我少写了一个$符号

closed 2019-04-15 by coolsnowwolf

qqj147258, 2019-04-24:
make[3]: Entering directory `/home/lcw/lede/package/lean/frpc'
touch /home/lcw/lede/build_dir/target-mips_24kc_musl/frp-0.26.0/.prepared_1fe666bb13d09e20992977d832c5155b_6664517399ebbbc92a37c5bb081b5c53_check
. /home/lcw/lede/include/shell.sh; gzip -dc /home/lcw/lede/dl/frp-0.26.0.tar.gz | tar -C /home/lcw/lede/build_dir/target-mips_24kc_musl/frp-0.26.0/.. -xf - 
[ ! -d ./src/ ] || cp -fpR ./src/. /home/lcw/lede/build_dir/target-mips_24kc_musl/frp-0.26.0

Applying ./patches/frpc.patch using plaintext: 
can't find file to patch at input line 3
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|--- a/package/lean/frpc/Makefile	2019-04-06 22:15:08.767294200 +0800
|+++ b/package/lean/frpc/Makefile	2019-04-14 22:12:43.732580300 +0800
--------------------------
No file to patch.  Skipping patch.
1 out of 1 hunk ignored
Patch failed!  Please fix ./patches/frpc.patch!
make[3]: *** [/home/lcw/lede/build_dir/target-mips_24kc_musl/frp-0.26.0/.prepared_1fe666bb13d09e20992977d832c5155b_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory `/home/lcw/lede/package/lean/frpc'
time: package/lean/frpc/compile#0.33#0.46#0.67
make[2]: *** [package/lean/frpc/compile] Error 2
make[2]: Leaving directory `/home/lcw/lede'
make[1]: *** [/home/lcw/lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/lcw/lede'
make: *** [world] 错误 2
报错了 大神帮忙看看怎么该

qqj147258, 2019-04-24:
https://gist.github.com/dongjuanyong/88eadaa3c94fa54fa56b66f188d46cab
这个大神的patch