coolsnowwolf/lede #1171: 用uClibc库编译在编译elfutils时报错，请dongjuanyong大佬指教该怎么办？先谢谢啦！

zxlhhyccc, 2019-04-14:
make[4]: Entering directory '/home/lin/le/lede-k3/build_dir/target-arm_cortex-a9_uClibc_eabi/elfutils-0.174'
make --no-print-directory all-recursive
Making all in config
make[6]: Nothing to be done for 'all'.
Making all in m4
make[6]: Nothing to be done for 'all'.
Making all in lib
CC xstrdup.o
CC xstrndup.o
CC xmalloc.o
CC next_prime.o
CC crc32.o
CC crc32_file.o
CC color.o
color.c: In function 'parse_opt':
color.c:135:5: error: passing argument 4 of 'argp_help' discards 'const' qualifier from pointer target type [-Werror=discarded-qualifiers]
program_invocation_short_name);
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from color.c:34:0:
/home/lin/le/lede-k3/staging_dir/target-arm_cortex-a9_uClibc_eabi/usr/include/argp.h:469:13: note: expected 'char *' but argument is of type 'const char *'
extern void argp_help (__const struct argp *__restrict __argp,
^~~~~~~~~
cc1: all warnings being treated as errors
Makefile:415: recipe for target 'color.o' failed
make[6]: *** [color.o] Error 1
Makefile:483: recipe for target 'all-recursive' failed
make[5]: *** [all-recursive] Error 1
Makefile:399: recipe for target 'all' failed
make[4]: *** [all] Error 2
make[4]: Leaving directory '/home/lin/le/lede-k3/build_dir/target-arm_cortex-a9_uClibc_eabi/elfutils-0.174'
Makefile:95: recipe for target '/home/lin/le/lede-k3/build_dir/target-arm_cortex-a9_uClibc_eabi/elfutils-0.174/.built' failed
make[3]: *** [/home/lin/le/lede-k3/build_dir/target-arm_cortex-a9_uClibc_eabi/elfutils-0.174/.built] Error 2
make[3]: Leaving directory '/home/lin/le/lede-k3/package/libs/elfutils'
time: package/libs/elfutils/compile#13.82#6.13#20.40
package/Makefile:107: recipe for target 'package/libs/elfutils/compile' failed
make[2]: *** [package/libs/elfutils/compile] Error 2
make[2]: Leaving directory '/home/lin/le/lede-k3'
package/Makefile:103: recipe for target '/home/lin/le/lede-k3/staging_dir/target-arm_cortex-a9_uClibc_eabi/stamp/.package_compile' failed
make[1]: *** [/home/lin/le/lede-k3/staging_dir/target-arm_cortex-a9_uClibc_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lin/le/lede-k3'
/home/lin/le/lede-k3/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

dongjuanyong, 2019-04-15:
第一我不是大佬，第二我没试过改c标准库。musl我用着觉得挺好，而且openwrt默认c库改成musl已经4年了，基本上所有package patch都是照着musl弄得，所以改成uClibc肯定免不了出问题
大概看了一眼，color.c里面对于argp_help的定义与uClibc库中argp.h里面的不一致，没仔细研究过，不一定对，至于怎么改我也不太清楚

zxlhhyccc, 2019-04-15:
K3屏幕天气的源码编译好像是在C库基础上的，所以需要在C库下编译，请问大佬你能研究一下C库这个报错吗？

coolsnowwolf, 2019-04-15:
这个不会再支持了，只会适配 musl

zxlhhyccc, 2019-04-15:
大雕大神，为啥不适配C库编译呢？

closed 2019-04-15 by coolsnowwolf