coolsnowwolf/lede #1177: v2ray 4.18.0编译报错

kanshudj, 2019-04-16:
全局科学上网，更新软件包，换架构编译都不行。
=============================================
Cleaning module download cache (golang/go#27455)
go: finding v2ray.com/core/.go_work/build latest
go: finding v2ray.com/core/.go_work latest
can't load package: package v2ray.com/core/.go_work/build: unknown import path "v2ray.com/core/.go_work/build": cannot find module providing package v2ray.com/core/.go_work/build

mv -f /home/xujian/lede/build_dir/target-aarch64_generic_musl/v2ray-core-4.18.0/.go_work/build/bin/linux_arm64/main /home/xujian/lede/build_dir/target-aarch64_generic_musl/v2ray-core-4.18.0/.go_work/build/bin/linux_arm64/v2ray
chmod +x /home/xujian/lede/build_dir/target-aarch64_generic_musl/v2ray-core-4.18.0/.go_work/build/bin/linux_arm64/v2ray && upx --ultra-brute /home/xujian/lede/build_dir/target-aarch64_generic_musl/v2ray-core-4.18.0/.go_work/build/bin/linux_arm64/v2ray
                       Ultimate Packer for eXecutables
                          Copyright (C) 1996 - 2013
UPX 3.91        Markus Oberhumer, Laszlo Molnar & John Reiser   Sep 30th 2013

        File size         Ratio      Format      Name
   --------------------   ------   -----------   -----------
upx: /home/xujian/lede/build_dir/target-aarch64_generic_musl/v2ray-core-4.18.0/.go_work/build/bin/linux_arm64/v2ray: UnknownExecutableFormatException

Packed 0 files.
Makefile:160: recipe for target '/home/xujian/lede/build_dir/target-aarch64_generic_musl/v2ray-core-4.18.0/.built' failed
make[3]: *** [/home/xujian/lede/build_dir/target-aarch64_generic_musl/v2ray-core-4.18.0/.built] Error 1
make[3]: Leaving directory '/home/xujian/lede/package/lean/v2ray'
time: package/lean/v2ray/compile#11.11#6.75#55.03
package/Makefile:107: recipe for target 'package/lean/v2ray/compile' failed
make[2]: *** [package/lean/v2ray/compile] Error 2
make[2]: Leaving directory '/home/xujian/lede'
package/Makefile:103: recipe for target '/home/xujian/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile' failed
make[1]: *** [/home/xujian/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/xujian/lede'
/home/xujian/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

coolsnowwolf, 2019-04-16:
upx 版本太低，不支持 arm64 导致的

coolsnowwolf, 2019-04-16:
UPX 不兼容 64bit elf，已禁用 upx 压缩

closed 2019-04-16 by coolsnowwolf