coolsnowwolf/lede #1181: 编译exfat-nofuse失败， No rule to make target

shelliao, 2019-04-16:
rm -f /home/shell/lede2/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5/.built
touch /home/shell/lede2/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5/.built_check
cat /dev/null > /home/shell/lede2/staging_dir/target-x86_64_musl/pkginfo/exfat-nofuse.symvers; for subdir in .; do cat /home/shell/lede2/staging_dir/target-x86_64_musl/pkginfo/*.symvers 2>/dev/null > /home/shell/lede2/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5/$subdir/Module.symvers; done
make -C "/home/shell/lede2/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.34" ARCH="x86" CROSS_COMPILE="x86_64-openwrt-linux-musl-" M="/home/shell/lede2/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5" CONFIG_EXFAT_FS=m modules
make[4]: Entering directory `/home/shell/lede2/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.34'
make[5]: *** No rule to make target `/home/shell/lede2/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5/exfat_core.o', needed by `/home/shell/lede2/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5/exfat.o'.  Stop.
make[4]: *** [_module_/home/shell/lede2/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5] Error 2
make[4]: Leaving directory `/home/shell/lede2/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.34'
make[3]: *** [/home/shell/lede2/build_dir/target-x86_64_musl/linux-x86_64/exfat-nofuse-2018-04-17-01c30ad5/.built] Error 2
make[3]: Leaving directory `/home/shell/lede2/feeds/packages/kernel/exfat-nofuse'
time: package/feeds/packages/exfat-nofuse/compile#0.73#0.18#16.26
make[2]: *** [package/feeds/packages/exfat-nofuse/compile] Error 2
make[2]: Leaving directory `/home/shell/lede2'
make[1]: *** [/home/shell/lede2/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/shell/lede2'
make: *** [world] Error 2
shell@OP-Compiler:~/lede2$

coolsnowwolf, 2019-04-16:
make clean

closed 2019-04-17 by coolsnowwolf

shelliao, 2019-04-18:
已经试过make clean，并且重新删掉所有文件夹重新下载编译仍然一样报错

ChengEast, 2020-02-14:
我也遇到同样的问题了，请教一下该如何处理呢？谢谢！！