coolsnowwolf/lede #1188: 编译coremark报错！

zxlhhyccc, 2019-04-18:
/lede/staging_dir/toolchain-arm_cortex-a9_gcc-7.4.0_glibc_eabi/arm-openwrt-linux-gnueabi/bin/../../../toolchain-arm_cortex-a9_gcc-7.4.0_glibc_eabi/lib/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
Makefile:86: recipe for target 'compile' failed
make[4]: *** [compile] Error 1
make[4]: Leaving directory '/home/lin/le/lede/build_dir/target-arm_cortex-a9_glibc_eabi/coremark-2019-01-31-2e9d6165'
Makefile:71: recipe for target '/home/lin/le/lede/build_dir/target-arm_cortex-a9_glibc_eabi/coremark-2019-01-31-2e9d6165/.built' failed
make[3]: *** [/home/lin/le/lede/build_dir/target-arm_cortex-a9_glibc_eabi/coremark-2019-01-31-2e9d6165/.built] Error 2
make[3]: Leaving directory '/home/lin/le/lede/package/lean/coremark'
time: package/lean/coremark/compile#3.53#2.46#18.05
package/Makefile:107: recipe for target 'package/lean/coremark/compile' failed
make[2]: *** [package/lean/coremark/compile] Error 2
make[2]: Leaving directory '/home/lin/le/lede'
package/Makefile:103: recipe for target '/home/lin/le/lede/staging_dir/target-arm_cortex-a9_glibc_eabi/stamp/.package_compile' failed
make[1]: *** [/home/lin/le/lede/staging_dir/target-arm_cortex-a9_glibc_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lin/le/lede'
/home/lin/le/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

coolsnowwolf, 2019-04-19:
musl 才能用

zxlhhyccc, 2019-04-19:
谢谢大佬！

zxlhhyccc, 2019-04-19:
大佬，能搞个coremark的LUCI吗？

closed 2019-04-19 by coolsnowwolf