coolsnowwolf/lede #1189: 刚刚更新源码，编译失败

jiushibushuo, 2019-04-18:
make[8]: Entering directory `/home/esir/lede/build_dir/target-x86_64_musl/glib-2.56.1/glib'
  CC       libglib_2_0_la-gbacktrace.lo
In file included from gbacktrace.c:35:0:
/home/esir/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/include/fortify/stdio.h: In function 'snprintf':
/home/esir/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/include/fortify/stdio.h:101:2: error: format not a string literal, argument types not checked [-Werror=format-nonliteral]
  return __orig_snprintf(__s, __n, __f, __builtin_va_arg_pack());
  ^~~~~~
/home/esir/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/include/fortify/stdio.h: In function 'sprintf':
/home/esir/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/include/fortify/stdio.h:110:3: error: format not a string literal, argument types not checked [-Werror=format-nonliteral]
   __r = __orig_snprintf(__s, __b, __f, __builtin_va_arg_pack());
   ^~~
/home/esir/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/include/fortify/stdio.h:114:3: error: format not a string literal, argument types not checked [-Werror=format-nonliteral]
   __r = __orig_sprintf(__s, __f, __builtin_va_arg_pack());
   ^~~
cc1: some warnings being treated as errors
make[8]: *** [libglib_2_0_la-gbacktrace.lo] Error 1
make[8]: Leaving directory `/home/esir/lede/build_dir/target-x86_64_musl/glib-2.56.1/glib'
make[7]: *** [all-recursive] Error 1
make[7]: Leaving directory `/home/esir/lede/build_dir/target-x86_64_musl/glib-2.56.1/glib'
make[6]: *** [all] Error 2
make[6]: Leaving directory `/home/esir/lede/build_dir/target-x86_64_musl/glib-2.56.1/glib'
make[5]: *** [all-recursive] Error 1
make[5]: Leaving directory `/home/esir/lede/build_dir/target-x86_64_musl/glib-2.56.1'
make[4]: *** [all] Error 2
make[4]: Leaving directory `/home/esir/lede/build_dir/target-x86_64_musl/glib-2.56.1'
make[3]: *** [/home/esir/lede/build_dir/target-x86_64_musl/glib-2.56.1/.built] Error 2
make[3]: Leaving directory `/home/esir/lede/feeds/packages/libs/glib2'
time: package/feeds/packages/glib2/compile#0.21#0.04#0.97
make[2]: *** [package/feeds/packages/glib2/compile] Error 2
make[2]: Leaving directory `/home/esir/lede'
make[1]: *** [/home/esir/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/esir/lede'
make: *** [world] Error 2

Troyshen94, 2019-04-18:
（新增了 upx autoconf automake libtool autopoint libelf-dev依赖）
 命令行输入 sudo apt-get update ，然后输入
sudo apt-get install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx autoconf automake libtool autopoint libelf-dev

jiushibushuo, 2019-04-19:
> （新增了 upx autoconf automake libtool autopoint libelf-dev依赖）
> 命令行输入 sudo apt-get update ，然后输入
> sudo apt-get install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev [...]

多谢，问题解决了！

closed 2019-04-19 by jiushibushuo