coolsnowwolf/lede #1192: 更新源码，v2ray编绎错误

Yangtian1971, 2019-04-19:
Cleaning module download cache (golang/go#27455)
go: finding v2ray.com/core/.go_work/build latest
go: finding v2ray.com/core/.go_work latest
can't load package: package v2ray.com/core/.go_work/build: unknown import path "v2ray.com/core/.go_work/build": cannot find module providing package v2ray.com/core/.go_work/build

Makefile:159: recipe for target '/home/yangchao/lede/build_dir/target-x86_64_musl/v2ray-core-4.18.0/.built' failed
make[3]: *** [/home/yangchao/lede/build_dir/target-x86_64_musl/v2ray-core-4.18.0/.built] Error 1
make[3]: Leaving directory '/home/yangchao/lede/package/lean/v2ray'
time: package/lean/v2ray/compile#17.09#6.95#158.95
package/Makefile:107: recipe for target 'package/lean/v2ray/compile' failed
make[2]: *** [package/lean/v2ray/compile] Error 2
make[2]: Leaving directory '/home/yangchao/lede'
package/Makefile:103: recipe for target '/home/yangchao/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/yangchao/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/yangchao/lede'
/home/yangchao/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

zxlhhyccc, 2019-04-19:
挂梯子吧

iotsys, 2019-04-19:
应该和梯子无关，我全局代理，编译 x86_64 也是这个情况

wangxin6, 2019-04-19:
在v2ray上出现类似问题：
make[3]: Entering directory '/home/parallels/lede/package/lean/v2ray'
rm -f /home/parallels/lede/build_dir/target-x86_64_musl/v2ray-core-4.18.0/.built
touch /home/parallels/lede/build_dir/target-x86_64_musl/v2ray-core-4.18.0/.built_check
( cd /home/parallels/lede/build_dir/target-x86_64_musl/v2ray-core-4.18.0/.go_work/build ; export GOPATH=/home/parallels/lede/build_dir/target-x86_64_musl/v2ray-core-4.18.0/.go_work/build GOCACHE=/home/parallels/lede/build_dir/target-x86_64_musl/v2ray-core-4.18.0/.go_work/cache GOTMPDIR=/home/parallels/lede/build_dir/target-x86_64_musl/v2ray-core-4.18.0/.go_work/tmp GOROOT_FINAL=/usr/lib/go-1.12 CC=x86_64-openwrt-linux-musl-gcc CXX=x86_64-openwrt-linux-musl-g++ 	GOOS=linux GOARCH=amd64 GO386=387 GOARM= GOMIPS= GOMIPS64= CGO_ENABLED=1 CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/parallels/lede/build_dir/target-x86_64_musl/v2ray-core-4.18.0:v2ray-core-4.18.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/parallels/lede/staging_dir/target-x86_64_musl/usr/include -I/home/parallels/lede/staging_dir/target-x86_64_musl/include -I/home/parallels/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/usr/include -I/home/parallels/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/include/fortify -I/home/parallels/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/include" CGO_CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -iremap/home/parallels/lede/build_dir/target-x86_64_musl/v2ray-core-4.18.0:v2ray-core-4.18.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" ; echo "Finding targets" ; targets=$(go list v2ray.com/core/main) ; for pattern in ; do targets=$(echo "$targets" | grep -v "$pattern") ; done ; echo ; if [ "" = 1 ]; then echo "Calling go generate" ; go generate -v -ldflags "-s -w" $targets ; echo ; fi ; if [ "" != 1 ]; then echo "Building targets" ; case amd64 in arm)             installsuffix="v" ;; mips|mipsle)     installsuffix="" ;; mips64|mips64le) installsuffix="" ;; esac ; trimpath="all=-trimpath=/home/parallels/lede/build_dir/target-x86_64_musl/v2ray-core-4.18.0/.go_work/build" ; ldflags="all=-linkmode external -extldflags '-L/home/parallels/lede/staging_dir/target-x86_64_musl/usr/lib -L/home/parallels/lede/staging_dir/target-x86_64_musl/lib -L/home/parallels/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/usr/lib -L/home/parallels/lede/staging_dir/toolchain-x86_64_gcc-7.4.0_musl/lib -znow -zrelro'" ; pkg_gcflags="" ; pkg_ldflags="" ; for def in ; do pkg_ldflags="$pkg_ldflags -X $def" ; done ; go install ${installsuffix:+-installsuffix $installsuffix} -gcflags "$trimpath" -asmflags "$trimpath" -ldflags "$ldflags" -v ${pkg_gcflags:+-gcflags "$pkg_gcflags"} ${pkg_ldflags:+-ldflags "$pkg_ldflags"} -ldflags "-s -w" $targets ; retval=$? ; echo ; if [ "$retval" -eq 0 ] && [ -z "$(find /home/parallels/lede/build_dir/target-x86_64_musl/v2ray-core-4.18.0/.go_work/build/bin -maxdepth 0 -type d \! -empty 2>/dev/null)" ]; then echo "No binaries were generated, consider adding GO_PKG_SOURCE_ONLY:=1 to Makefile" ; echo ; fi ; echo "Cleaning module download cache (golang/go#27455)" ; go clean -modcache ; echo ; fi ; exit $retval ; )
Finding targets
go: finding google.golang.org/genproto v0.0.0-20180817151627-c66870c02cf8
go: finding google.golang.org/genproto v0.0.0-20180831171423-11092d34479b
go: google.golang.org/genproto@v0.0.0-20180817151627-c66870c02cf8: git fetch -f https://github.com/google/go-genproto refs/heads/*:refs/heads/* refs/tags/*:refs/tags/* in /home/parallels/lede/build_dir/target-x86_64_musl/v2ray-core-4.18.0/.go_work/build/pkg/mod/cache/vcs/6310ae0a635ce9b4d0073713c68a8462977a85bc4f6c9d1fd99854a36fcab1f0: exit status 128:
	error: RPC failed; curl 56 GnuTLS recv error (-54): Error in the pull function.
	fatal: The remote end hung up unexpectedly
	fatal: early EOF
	fatal: index-pack failed
go: google.golang.org/genproto@v0.0.0-20180831171423-11092d34479b: unknown revision 11092d34479b
go: error loading module requirements

coolsnowwolf, 2019-04-19:
error: RPC failed; curl 56 GnuTLS recv error (-54): Error in the pull function.

被墙了

wangxin6, 2019-04-19:
路由器设置全局🍅搞定。谢谢 @coolsnowwolf

lllrrr, 2019-04-20:
服务器不在国内的也出现报错。ping golang.org能通

closed 2019-04-20 by coolsnowwolf