coolsnowwolf/lede #1198: v2ray编译报错-“can't load package: “；”unknown import path“

wgjwillis, 2019-04-20:
环境：
vmware 下的ubuntu 18.10 64bit，全局vpn，台湾的梯子。

报错代码：
`Cleaning module download cache (golang/go#27455)
https://v2ray.com?go-get=1go: finding v2ray.com/core/.go_work/build latest
go: finding v2ray.com/core/.go_work latest
can't load package: package v2ray.com/core/.go_work/build: unknown import path "v2ray.com/core/.go_work/build": cannot find module providing package v2ray.com/core/.go_work/build

make[3]: *** [Makefile:163: /home/willis/lede/build_dir/target-mips_24kc_musl/v2ray-core-4.18.0/.built] Error 1
make[3]: Leaving directory '/home/willis/lede/package/lean/v2ray'
time: package/lean/v2ray/compile#13.64#13.25#265.47
make[2]: *** [package/Makefile:109: package/lean/v2ray/compile] Error 2
make[2]: Leaving directory '/home/willis/lede'
make[1]: *** [package/Makefile:103: /home/willis/lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/willis/lede'
make: *** [/home/willis/lede/include/toplevel.mk:218：world] 错误 2
`
vm的ubuntu 中浏览器能访问：v2ray.com/core/.go_work/build 但是跳转到https://godoc.org/github.com/v2ray/v2ray-core

请问如何解决？

zxlhhyccc, 2019-04-20:
挂梯子。。。。。

wgjwillis, 2019-04-20:
> 挂梯子。。。。。

挂了的，系统中用的vpn，全局，台湾的梯子。

zxlhhyccc, 2019-04-20:
路由里设置，用SSR挂全局

closed 2019-04-20 by coolsnowwolf

wgjwillis, 2019-04-20:
是的，我主路由用haporxy做了ssr的负载均衡。当编译到用go 去get的时候，节点会变，导致go get 失败。固定一个ssr节点的全局代理，或者vpn就没有问题了。