coolsnowwolf/lede #1207: Newifi D2/B70 有时候会出现Wifi消失

ddvcx, 2019-04-21:
可能几天或者几个星期出现一次
就是Wifi信号完全找不到，2.4G和5G都找不到
有时候过个几十分钟Wifi自动恢复，有时候得重启
这是系统日志
Fri Apr 19 11:40:56 2019 daemon.notice hostapd: wlan0: interface state COUNTRY_UPDATE->ENABLED
Fri Apr 19 11:40:56 2019 daemon.notice hostapd: wlan0: AP-ENABLED
Fri Apr 19 11:40:56 2019 daemon.info procd: - init complete -
Fri Apr 19 11:40:56 2019 daemon.notice netifd: Network device 'wlan0' link is up
Fri Apr 19 11:40:57 2019 kern.info kernel: [   26.622321] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
Fri Apr 19 11:40:57 2019 kern.info kernel: [   26.628839] br-lan: port 3(wlan1) entered blocking state
Fri Apr 19 11:40:57 2019 kern.info kernel: [   26.634178] br-lan: port 3(wlan1) entered forwarding state
Fri Apr 19 11:40:57 2019 daemon.notice hostapd: wlan1: interface state HT_SCAN->ENABLED
Fri Apr 19 11:40:57 2019 daemon.notice hostapd: wlan1: AP-ENABLED
Fri Apr 19 11:40:57 2019 daemon.notice netifd: Network device 'wlan1' link is up
Fri Apr 19 11:40:57 2019 daemon.info hostapd: wlan0: STA 78:da:07:6d:d8:df IEEE 802.11: authenticated
Fri Apr 19 11:40:57 2019 daemon.info hostapd: wlan0: STA 78:da:07:6d:d8:df IEEE 802.11: associated (aid 1)
Fri Apr 19 11:40:57 2019 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 78:da:07:6d:d8:df
Fri Apr 19 11:40:57 2019 daemon.info hostapd: wlan0: STA 78:da:07:6d:d8:df WPA: pairwise key handshake completed (RSN)
Fri Apr 19 11:40:57 2019 daemon.notice netifd: wan (1494): udhcpc: sending discover
Fri Apr 19 11:41:00 2019 daemon.notice netifd: wan (1494): udhcpc: sending discover
Fri Apr 19 11:41:10 2019 daemon.info hostapd: wlan0: STA 20:ab:37:38:f2:48 IEEE 802.11: authenticated
Fri Apr 19 11:41:10 2019 daemon.info hostapd: wlan0: STA 20:ab:37:38:f2:48 IEEE 802.11: associated (aid 2)
Fri Apr 19 11:41:10 2019 daemon.notice hostapd: wlan0: AP-STA-CONNECTED 20:ab:37:38:f2:48
Fri Apr 19 11:41:10 2019 daemon.info hostapd: wlan0: STA 20:ab:37:38:f2:48 WPA: pairwise key handshake completed (RSN)
Fri Apr 19 11:41:54 2019 daemon.info hostapd: wlan0: STA ac:5f:3e:b4:46:91 IEEE 802.11: authenticated
Fri Apr 19 11:41:54 2019 daemon.info hostapd: wlan0: STA ac:5f:3e:b4:46:91 IEEE 802.11: associated (aid 3)
Fri Apr 19 11:41:54 2019 daemon.notice hostapd: wlan0: AP-STA-CONNECTED ac:5f:3e:b4:46:91
Fri Apr 19 11:41:54 2019 daemon.info hostapd: wlan0: STA ac:5f:3e:b4:46:91 WPA: pairwise key handshake completed (RSN)
Fri Apr 19 11:42:38 2019 daemon.info hostapd: wlan0: STA fc:fc:48:87:3d:db IEEE 802.11: authenticated
Fri Apr 19 11:42:38 2019 daemon.info hostapd: wlan0: STA fc:fc:48:87:3d:db IEEE 802.11: associated (aid 4)
Fri Apr 19 11:42:38 2019 daemon.notice hostapd: wlan0: AP-STA-CONNECTED fc:fc:48:87:3d:db
Fri Apr 19 11:42:38 2019 daemon.info hostapd: wlan0: STA fc:fc:48:87:3d:db WPA: pairwise key handshake completed (RSN)
Fri Apr 19 11:43:11 2019 daemon.info hostapd: wlan1: STA 34:13:e8:99:4e:31 IEEE 802.11: authenticated
Fri Apr 19 11:43:11 2019 daemon.info hostapd: wlan1: STA 34:13:e8:99:4e:31 IEEE 802.11: associated (aid 1)
Fri Apr 19 11:43:11 2019 daemon.notice hostapd: wlan1: STA-OPMODE-SMPS-MODE-CHANGED 34:13:e8:99:4e:31 off
Fri Apr 19 11:43:11 2019 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 34:13:e8:99:4e:31
Fri Apr 19 11:43:11 2019 daemon.info hostapd: wlan1: STA 34:13:e8:99:4e:31 WPA: pairwise key handshake completed (RSN)
Fri Apr 19 11:43:59 2019 daemon.err uhttpd[1378]: luci: accepted login on / for root from 192.168.2.11

hcym, 2019-04-22:
硬件故障的话就别折腾了，我只有一个新3，买来就刷了op，基本是官方op，大雕的不太用，除了v2会卡死，基本没出过问题

ddvcx, 2019-04-22:
> 硬件故障的话就别折腾了，我只有一个新3，买来就刷了op，基本是官方op，大雕的不太用，除了v2会卡死，基本没出过问题

硬件故障也不可能两个路由器都一样的故障啊
而且这个故障出现是随机的，有可能10天半个月不出来，也有可能2 3天就出现

closed 2019-09-29 by coolsnowwolf

hunhun010164, 2020-05-24:
这个路由器超级烂,我刷了官方的openwrt 19.07.2,几个小时就重启一次,超级超级超级垃圾!!!联想的东西果然都是垃圾!