coolsnowwolf/lede #1209: 请教各位大神

ziqqqq, 2019-04-21:
net/ipv4/.tcp_bbr.o.cmd:68: *** missing separator.  Stop.
scripts/Makefile.build:585: recipe for target 'net/ipv4' failed
make[6]: *** [net/ipv4] Error 2
Makefile:1039: recipe for target 'net' failed
make[5]: *** [net] Error 2
make[5]: Leaving directory '/home/ll/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.111'
Makefile:26: recipe for target '/home/ll/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.111/.modules' failed
make[4]: *** [/home/ll/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.111/.modules] Error 2
make[4]: Leaving directory '/home/ll/lede/target/linux/ramips'
Makefile:13: recipe for target 'compile' failed
make[3]: *** [compile] Error 2
make[3]: Leaving directory '/home/ll/lede/target/linux'
time: target/linux/compile#3.51#68.14#97.90
target/Makefile:23: recipe for target 'target/linux/compile' failed
make[2]: *** [target/linux/compile] Error 2
make[2]: Leaving directory '/home/ll/lede'
target/Makefile:18: recipe for target '/home/ll/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_compile' failed
make[1]: *** [/home/ll/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/ll/lede'
/home/ll/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

hcym, 2019-04-22:
只看到bbr，其它都不懂了，

lan542662, 2019-04-22:
删了全局重来。

wangxin6, 2019-04-22:
路由器上设置母鸡IP全局翻墙很重要。

closed 2019-04-27 by coolsnowwolf