coolsnowwolf/lede #1210: newifi d2 编译错误

barton0426, 2019-04-22:
错误代码是
go tool dist: FAILED: /mnt/lede/build_dir/hostpkg/go-1.12.1/pkg/tool/linux_amd64/go_bootstrap install -gcflags=all= -ldflags=all= -a -i cmd/asm cmd/cgo cmd/compile cmd/link: exit status 2

barton0426, 2019-04-22:
更换了asus 1300uhp 依然报错 golang编译问题

runtime.(*mheap).alloc(0x16a57c0, 0x1, 0xc000010012, 0x457470)
        /mnt/lede/staging_dir/hostpkg/lib/go-cross/src/runtime/mheap.go:1047 +0x8a fp=0xc01081b0b0 sp=0xc01081b060 pc=0x423faa
runtime.(*mcentral).grow(0x16a5fc0, 0x0)
        /mnt/lede/staging_dir/hostpkg/lib/go-cross/src/runtime/mcentral.go:256 +0x95 fp=0xc01081b0f8 sp=0xc01081b0b0 pc=0x416ee5
runtime.(*mcentral).cacheSpan(0x16a5fc0, 0xc01081b188)
        /mnt/lede/staging_dir/hostpkg/lib/go-cross/src/runtime/mcentral.go:106 +0x2ff fp=0xc01081b158 sp=0xc01081b0f8 pc=0x4169ef
runtime.(*mcache).refill(0x7f00c9bd5008, 0x12)
        /mnt/lede/staging_dir/hostpkg/lib/go-cross/src/runtime/mcache.go:135 +0x86 fp=0xc01081b178 sp=0xc01081b158 pc=0x416486
runtime.(*mcache).nextFree(0x7f00c9bd5008, 0xc04bf51d12, 0xc010811981, 0xc0000ea0b8, 0x28)
        /mnt/lede/staging_dir/hostpkg/lib/go-cross/src/runtime/malloc.go:786 +0x88 fp=0xc01081b1b0 sp=0xc01081b178 pc=0x40b3c8
runtime.mallocgc(0x80, 0xdfbae0, 0x1636201, 0x1636220)
        /mnt/lede/staging_dir/hostpkg/lib/go-cross/src/runtime/malloc.go:939 +0x76e fp=0xc01081b250 sp=0xc01081b1b0 pc=0x40bcde
runtime.growslice(0xdfbae0, 0xc004ab8b40, 0x4, 0x4, 0x5, 0xc004ab8b40, 0x2, 0x4)
        /mnt/lede/staging_dir/hostpkg/lib/go-cross/src/runtime/slice.go:181 +0x1e6 fp=0xc01081b2b8 sp=0xc01081b250 pc=0x4429e6
cmd/internal/obj.(*FuncInfo).AddInlMark(...)
        /mnt/lede/staging_dir/hostpkg/lib/go-cross/src/cmd/internal/obj/link.go:429
cmd/compile/internal/gc.genssa(0xc04bd51760, 0xc04be9bf10)
        /mnt/lede/staging_dir/hostpkg/lib/go-cross/src/cmd/compile/internal/gc/ssa.go:5179 +0x93b fp=0xc01081b6e8 sp=0xc01081b2b8 pc=0xcd222b
cmd/compile/internal/gc.compileSSA(0xc00b5362c0, 0x0)
        /mnt/lede/staging_dir/hostpkg/lib/go-cross/src/cmd/compile/internal/gc/pgen.go:309 +0x350 fp=0xc01081b798 sp=0xc01081b6e8 pc=0xc71b90
cmd/compile/internal/gc.compile(0xc00b5362c0)
        /mnt/lede/staging_dir/hostpkg/lib/go-cross/src/cmd/compile/internal/gc/pgen.go:278 +0x5b1 fp=0xc01081b838 sp=0xc01081b798 pc=0xc717b1
cmd/compile/internal/gc.funccompile(0xc00b5362c0)
        /mnt/lede/staging_dir/hostpkg/lib/go-cross/src/cmd/compile/internal/gc/pgen.go:221 +0xc2 fp=0xc01081b890 sp=0xc01081b838 pc=0xc710f2
cmd/compile/internal/gc.Main(0xe76690)
        /mnt/lede/staging_dir/hostpkg/lib/go-cross/src/cmd/compile/internal/gc/main.go:665 +0x3059 fp=0xc01081bf20 sp=0xc01081b890 pc=0xc48ce9
main.main()
        /mnt/lede/staging_dir/hostpkg/lib/go-cross/src/cmd/compile/main.go:51 +0xad fp=0xc01081bf98 sp=0xc01081bf20 pc=0xd83d5d
runtime.main()
        /mnt/lede/staging_dir/hostpkg/lib/go-cross/src/runtime/proc.go:200 +0x20c fp=0xc01081bfe0 sp=0xc01081bf98 pc=0x42e58c
runtime.goexit()
        /mnt/lede/staging_dir/hostpkg/lib/go-cross/src/runtime/asm_amd64.s:1337 +0x1 fp=0xc01081bfe8 sp=0xc01081bfe0 pc=0x45aec1
go tool dist: FAILED: /mnt/lede/build_dir/hostpkg/go-1.12.1/pkg/tool/linux_amd64/go_bootstrap install -gcflags=all= -ldflags=all= -a -i cmd/asm cmd/cgo cmd/compile cmd/link: exit status 2
make[3]: *** [Makefile:275: /mnt/lede/build_dir/hostpkg/go-1.12.1/.built] Error 2
make[3]: Leaving directory '/mnt/lede/feeds/packages/lang/golang/golang'
time: package/feeds/packages/golang/host-compile#256.46#38.69#501.20
make[2]: *** [package/Makefile:107: package/feeds/packages/golang/host/compile] Error 2

hcym, 2019-04-22:
梯子问题吧？我中午刚刚编译的n1没有什么事，kcp和V2ray都要用这个go的

barton0426, 2019-04-23:
> 梯子问题吧？我中午刚刚编译的n1没有什么事，kcp和V2ray都要用这个go的

vps上弄得 网络肯定不是问题 我怀疑swap不够大。。 不过我的20g vps现在好像应付不了这个玩意了。。。

hcym, 2019-04-23:
本机的话，16G空间基本够用一次，因为我习惯挂shm内存上编译，，完事看内存还有4G不到点。32G就轻松自在了，多次编译也不会爆。

closed 2019-04-24 by coolsnowwolf