coolsnowwolf/lede #1213: 这段时间编译X86老是出现如下错误，请指正

dongfangy2011, 2019-04-22:
make[4]: Entering directory '/home/df/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.34'
make[5]: *** No rule to make target 'tools/objtool/objtool', needed by '/home/df/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0/xt_FULLCONENAT.o'.  Stop.
Makefile:1523: recipe for target '_module_/home/df/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0' failed
make[4]: *** [_module_/home/df/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0] Error 2
make[4]: Leaving directory '/home/df/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.34'
Makefile:64: recipe for target '/home/df/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0/.built' failed
make[3]: *** [/home/df/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2018-12-15-d4daedd0/.built] Error 2
make[3]: Leaving directory '/home/df/lede/package/lean/openwrt-fullconenat'
time: package/lean/openwrt-fullconenat/compile#1.75#1.65#3.21
package/Makefile:107: recipe for target 'package/lean/openwrt-fullconenat/compile' failed
make[2]: *** [package/lean/openwrt-fullconenat/compile] Error 2
make[2]: Leaving directory '/home/df/lede'
package/Makefile:103: recipe for target '/home/df/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/df/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/df/lede'
/home/df/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

lan542662, 2019-04-22:
根据经验，删了重编译，还有保证分区足够大。

coolsnowwolf, 2019-04-23:
sudo apt-get -y install build-essential asciidoc binutils bzip2 gawk gettext git libncurses5-dev libz-dev patch unzip zlib1g-dev lib32gcc1 libc6-dev-i386 subversion flex uglifyjs git-core gcc-multilib p7zip p7zip-full msmtp libssl-dev texinfo libglib2.0-dev xmlto qemu-utils upx libelf-dev autoconf automake libtool autopoint

closed 2019-04-23 by coolsnowwolf

Mokee-nmy, 2020-01-02:
可以尝试：sudo apt-get update
如果还是不行，可以更换软件源，比如换成清华的或者使用系统自带的源

jzhou404, 2020-01-13:
谢谢，coolsnowwolf, 解决了我的 recipe for target 'package/lean/openwrt-fullconenat/compile' failed

问题