coolsnowwolf/lede #1214: SSR-Plus三合一连接v2节点问题

Retrochet, 2019-04-22:
mt7621的设备，加了fpu模拟功能编译的固件，输入正确的v2节点参数后，日志永远是
```
……
2019-04-22 20:04:55 Current server error, try to switch another server.
2019-04-22 19:50:59 Current server error, try to switch another server.
2019-04-22 19:40:54 V2Ray 4.18.0 (Po) Lean's custom build - Release 5 Started!
```
无法访问外网。v2ray和v2ray-core都分别编译试过了，请问是缺少依赖还是什么问题？

Retrochet, 2019-04-23:
追加：进入ssh之后使用v2ray -test指令得到
···
main: failed to load config:  > v2ray.com/core/main/confloader/external: config file not readable > open : no such file or directory
···

Retrochet, 2019-04-23:
补充配置文件测试：
```
/usr/bin/v2ray/v2ray -test -config=/tmp/etc/v2-ssr-ret
cp.json
V2Ray 4.18.0 (Po) Lean's custom build - Release 5
A unified platform for anti-censorship.
Configuration OK.
```

zxlhhyccc, 2019-04-23:
> 补充配置文件测试：
> 
> ```
> /usr/bin/v2ray/v2ray -test -config=/tmp/etc/v2-ssr-ret
> cp.json
> V2Ray 4.18.0 (Po) Lean's custom build - Release 5
> A unified platform for anti-censorship.
> Configuration OK.
> ```

应该是没编译v2ctl

Retrochet, 2019-04-23:
>应该是没编译v2ctl  
的确没有编译。无论是否为完整的v2ray-core都是必要的依赖吗？

hcym, 2019-04-24:
core是完整版，全有的，而且大雕码不依赖这个，看看其它问题

Retrochet, 2019-04-24:
目前能想到的只有lua-cjson和filetransfer没有编译进去，不过这两个事以前v2raypro的依赖。如何排查问题？

coolsnowwolf, 2019-04-25:
不需要的，配置参数给一下

Retrochet, 2019-04-25:
>不需要的，配置参数给一下

/tmp/etc/

```
{
   "outbound": {
     "settings": {
       "vnext": [
         {
           "port": 443,
           "users": [
             {
               "id": "xxxxxxxxxxxxxxxxxxxxxxxx",
               "alterId": 4,
               "security": "chacha20-poly1305"
             }
           ],
           "address": "www.mydomain.com"
         }
       ]
     },
     "mux": {
       "enabled": true
     },
     "protocol": "vmess",
     "streamSettings": {
       "network": "ws",
       "tlsSettings": {
         "allowInsecure": false
       },
       "wsSettings": {
         "path": "\/v2ray"
       },
       "security": "none"
     }
   },
   "log": {
     "loglevel": "warning"
   },
   "outboundDetour": [
     {
       "settings": {
         "keep": ""
       },
       "protocol": "freedom",
       "tag": "direct"
     }
   ],
   "inbound": {
     "sniffing": {
       "enabled": true,
       "destOverride": [
         "http",
         "tls"
       ]
     },
     "port": "1080",
     "protocol": "dokodemo-door",
     "settings": {
       "network": "tcp",
       "followRedirect": true
     }
   }
 }
```

 我注意到path多了个\，但在luci上删除/之后path一行就变成了
```
……
      "path": "v2ray"
……
```

Retrochet, 2019-04-25:
补充：  
一个现象：现在在开启ssr-plus的状态下重启后，无论luci界面ssr-plus是否开启，都无法联网。在luci界面停用v2节点后重启恢复正常上网，此时/tmp/etc下相关json配置文件自动消失  
编译相关：官方openwrt主分支，根据luci-app-ssr-plus的makefile，编译出的固件在包管理器中确认其依赖均已安装：
luci-app-ssr-plus
shadowsocksr-libev-alt
ipset
ip-full
iptables-mod-tproxy
dnsmasq-full
coreutils
coreutils-base64
bash
pdnsd-alt
wget
v2ray / v2ray-core  （由于仅需v2故只勾选了此可选项）

zxlhhyccc, 2019-05-01:
> mt7621的设备，加了fpu模拟功能编译的固件，输入正确的v2节点参数后，日志永远是
> 
> ```
> ……
> 2019-04-22 20:00:55 Current server error, try to switch another server.
> 2019-04-22 19:50:59 Current server error, try to switch another server.
> 2019-04-22 19:40:54 V2Ray 4.18.0 (Po) Lean's custom build - Release 5 Started!
> ```
> 
> 无法访问外网。v2ray和v2ray-core都分别编译试过了，请问是缺少依赖还是什么问题？

FPU模拟你是怎么加进去的？

AkinoMaple, 2019-05-02:
\是转义符 但是用了引号应该不需要对/了qwq

Retrochet, 2019-05-02:
>\是转义符 但是用了引号应该不需要对/了qwq

去掉/之后依然是Current server error, try to switch another server.

Retrochet, 2019-05-02:
补充：更新至5月1日的master后重新编译，问题依然存在。  
所有安装的包如下  
aria2	1.34.0-6	-	-	
base-files	197-r9916-a9190ee3a4	-	-	
bash	5.0-2	-	-	
block-mount	2019-03-28-ff1ded63-4	-	-	
busybox	1.30.1-2	-	-	
ca-bundle	20190110-1	-	-	
ca-certificates	20190110-1	-	-	
coreutils	8.30-1	-	-	
coreutils-base64	8.30-1	-	-	
dnsmasq-full	2.80-11	-	-	
dropbear	2019.78-2	-	-	
etherwake	1.09-4	-	-	
firewall	2019-01-02-70f8785b-2	-	-	
frpc	0.26.0-1	-	-	
fstools	2019-03-28-ff1ded63-4	-	-	
fwtool	1	-	-	
hostapd-common	2018-12-02-c2c6c01b-6	-	-	
https_dns_proxy	2018-04-23-3	-	-	
ip-full	5.0.0-1	-	-	
ip6tables	1.8.2-3	-	-	
ipset	7.1-3	-	-	
iptables	1.8.2-3	-	-	
iptables-mod-conntrack-extra	1.8.2-3	-	-	
iptables-mod-fullconenat	2018-12-15-d4daedd0-1	-	-	
iptables-mod-ipopt	1.8.2-3	-	-	
iptables-mod-tproxy	1.8.2-3	-	-	
iw	5.0.1-1	-	-	
iwinfo	2019-03-28-c2cfe9d9-1	-	-	
jshn	2019-02-27-eeef7b50-1	-	-	
jsonfilter	2018-02-04-c7e938d6-1	-	-	
kernel	4.14.113-1-ce0a75f7…	-	-	
kmod-cfg80211	4.14.113+4.19.32-1-2	-	-	
kmod-crypto-crc32c	4.14.113-1	-	-	
kmod-crypto-hash	4.14.113-1	-	-	
kmod-fs-ext4	4.14.113-1	-	-	
kmod-fs-vfat	4.14.113-1	-	-	
kmod-fuse	4.14.113-1	-	-	
kmod-gpio-button-hotplug	4.14.113-2	-	-	
kmod-ifb	4.14.113-1	-	-	
kmod-ip6tables	4.14.113-1	-	-	
kmod-ipt-conntrack	4.14.113-1	-	-	
kmod-ipt-conntrack-extra	4.14.113-1	-	-	
kmod-ipt-core	4.14.113-1	-	-	
kmod-ipt-fullconenat	4.14.113+2018-12-15-d4daedd0-1	-	-	
kmod-ipt-ipopt	4.14.113-1	-	-	
kmod-ipt-ipset	4.14.113-1	-	-	
kmod-ipt-nat	4.14.113-1	-	-	
kmod-ipt-offload	4.14.113-1	-	-	
kmod-ipt-raw	4.14.113-1	-	-	
kmod-ipt-tproxy	4.14.113-1	-	-	
kmod-leds-gpio	4.14.113-1	-	-	
kmod-lib-crc-ccitt	4.14.113-1	-	-	
kmod-lib-crc16	4.14.113-1	-	-	
kmod-mac80211	4.14.113+4.19.32-1-2	-	-	
kmod-mmc	4.14.113-1	-	-	
kmod-mt76-core	4.14.113+2019-03-27-a11b6734-1	-	-	
kmod-mt7603	4.14.113+2019-03-27-a11b6734-1	-	-	
kmod-mt76x02-common	4.14.113+2019-03-27-a11b6734-1	-	-	
kmod-mt76x2	4.14.113+2019-03-27-a11b6734-1	-	-	
kmod-mt76x2-common	4.14.113+2019-03-27-a11b6734-1	-	-	
kmod-nf-conntrack	4.14.113-1	-	-	
kmod-nf-conntrack-netlink	4.14.113-1	-	-	
kmod-nf-conntrack6	4.14.113-1	-	-	
kmod-nf-flow	4.14.113-1	-	-	
kmod-nf-ipt	4.14.113-1	-	-	
kmod-nf-ipt6	4.14.113-1	-	-	
kmod-nf-nat	4.14.113-1	-	-	
kmod-nf-reject	4.14.113-1	-	-	
[... 126 lines cut ...]
procd	2019-04-26-cfaed563-2	-	-	
rpcd	2018-11-28-3aa81d0d-1	-	-	
rpcd-mod-rrdns	20170710	-	-	
samba36-server	3.6.25-14	-	-	
shadowsocksr-libev-alt	2.5.6-3	-	-	
sqm-scripts
swconfig	12	-	-	
tc	5.0.0-1	-	-	
terminfo	6.1-4	-	-	
ubi-utils	2.0.2-1	-	-	
ubox	2019-04-07-5130fa4d-1	-	-	
ubus	2018-10-06-221ce7e7-1	-	-	
ubusd	2018-10-06-221ce7e7-1	-	-	
uci	2018-08-11-4c8b4d6e-2	-	-	
uclibcxx	0.2.4-3	-	-	
uclient-fetch	2018-11-24-3ba74ebc-1	-	-	
uhttpd	2018-11-28-cdfc902a-3	-	-	
usign	2015-07-04-ef641914-1	-	-	
v2ray	v4.19	-	-	
vlmcsd	svn1112-1	-	-	
vsftpd-alt	3.0.3-6	-	-	
wget	1.20.3-1	-	-	
wireless-regdb	2017-10-20-4343d359	-	-	
wpad	2018-12-02-c2c6c01b-6	-	-	
zlib	1.2.11-2

Retrochet, 2019-05-02:
> 不需要的，配置参数给一下

closed 2019-09-29 by coolsnowwolf