coolsnowwolf/lede #1221: make kernel_menuconfig 出错

wangxin6, 2019-04-24:
make kernel_menuconfig 出错：

make[1]: Entering directory '/home/parallels/lede/target/linux'
make[2]: Entering directory '/home/parallels/lede/target/linux/x86'
rm -f /home/parallels/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.36/.config.prev
rm -f /home/parallels/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.36/.configured
/home/parallels/lede/scripts/kconfig.pl  + /home/parallels/lede/target/linux/generic/config-4.19 /home/parallels/lede/target/linux/x86/config-4.19 > /home/parallels/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.36/.config
export MAKEFLAGS= ;make -C /home/parallels/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.36 HOSTCFLAGS="-O2 -I/home/parallels/lede/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Wed Apr 24 06:06:46 2019" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/home/parallels/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KERNELRELEASE=4.19.36 CC="x86_64-openwrt-linux-musl-gcc"  menuconfig
make[3]: Entering directory '/home/parallels/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.36'
/bin/sh: 1: x86_64-openwrt-linux-musl-gcc: not found
/bin/sh: 1: x86_64-openwrt-linux-musl-gcc: not found
--define-variable argument does not have a value for the variable
--define-variable argument does not have a value for the variable
scripts/kconfig/mconf  Kconfig
./scripts/gcc-version.sh: 26: ./scripts/gcc-version.sh: x86_64-openwrt-linux-musl-gcc: not found
./scripts/gcc-version.sh: 27: ./scripts/gcc-version.sh: x86_64-openwrt-linux-musl-gcc: not found
./scripts/gcc-version.sh: 29: ./scripts/gcc-version.sh: x86_64-openwrt-linux-musl-gcc: not found
./scripts/gcc-version.sh: 26: ./scripts/gcc-version.sh: x86_64-openwrt-linux-musl-gcc: not found
./scripts/gcc-version.sh: 27: ./scripts/gcc-version.sh: x86_64-openwrt-linux-musl-gcc: not found
./scripts/gcc-version.sh: 29: ./scripts/gcc-version.sh: x86_64-openwrt-linux-musl-gcc: not found
init/Kconfig:17: syntax error
init/Kconfig:16: invalid option
./scripts/clang-version.sh: 15: ./scripts/clang-version.sh: x86_64-openwrt-linux-musl-gcc: not found
./scripts/gcc-plugin.sh: 11: ./scripts/gcc-plugin.sh: x86_64-openwrt-linux-musl-gcc: not found
net/sched/Kconfig:44: warning: menuconfig statement without prompt
scripts/kconfig/Makefile:29: recipe for target 'menuconfig' failed
make[4]: *** [menuconfig] Error 1
Makefile:544: recipe for target 'menuconfig' failed
make[3]: *** [menuconfig] Error 2
make[3]: Leaving directory '/home/parallels/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-4.19.36'
Makefile:30: recipe for target 'menuconfig' failed
make[2]: *** [menuconfig] Error 2
make[2]: Leaving directory '/home/parallels/lede/target/linux/x86'
Makefile:13: recipe for target 'menuconfig' failed
make[1]: *** [menuconfig] Error 2
make[1]: Leaving directory '/home/parallels/lede/target/linux'
/home/parallels/lede/include/toplevel.mk:161: recipe for target 'kernel_menuconfig' failed
make: *** [kernel_menuconfig] Error 2

wangxin6, 2019-04-24:
已尝试使用make clean无效

wangxin6, 2019-04-25:
git clone重新下载源码，不做任何配置修改，运行make kernel_menuconfig。错误依旧

coolsnowwolf, 2019-04-25:
先编译，才能配置

wangxin6, 2019-04-25:
多谢。

closed 2019-04-25 by wangxin6