coolsnowwolf/lede #1228: SSR Plus使用SS客户端，但Socks5使用的是ssr-local导致Socks5无效

kedzlcy, 2019-04-26:
节点类型选择Shadowsocks New Version，加密方式选择 acs-256-gcm。

在路由器中ps显示下面进程。

/usr/bin/ss-redir -c /var/etc/shadowsocksr.json -f /var/run/ssr-retcp_1.pid
/usr/bin/ss-redir -c /var/etc/shadowsocksr.json -f /var/run/ssr-retcp_2.pid
/usr/bin/ssr-local -c /var/etc/shadowsocksr_s.json -u -l 1080 -b 0.0.0.0 -f /var/run/ssr-local.pid

杀死ssr-local 后启动下面脚本后SOCKS5服务正常运行。
/usr/bin/ss-local -c /var/etc/shadowsocksr.json -u -l 1080 -b 0.0.0.0 -f /var/run/ss-local.pid

coolsnowwolf, 2019-04-28:
嗯，这里确实没考虑，后面修复吧

kedzlcy, 2019-04-29:
SOCKS5服务在SSR PLUS客户端属于禁用的状态下，仍然会开机自启动。
在kill掉ssr-local的情况下，修改SSR PLUS里的客户端，服务器节点，访问控制，高级设置，服务端都会重新拉起ssr-local进程。

SOCKS5代理会出现无法选择服务器，但服务属于启动状态的情况。

希望在SSR Plus的服务端加入SS Server的选项，支持 acs-256-gcm。

coolsnowwolf, 2019-04-30:
SOCKS5服务在SSR PLUS客户端属于禁用的状态下，仍然会开机自启动

这个是有意这样设计的，因为有些人不需要透明代理啊，就需要 socks5

505575853, 2019-05-05:
https://github.com/505575853/lede/tree/R9/package/lean/luci-app-ssr-plus

closed 2019-09-29 by coolsnowwolf