coolsnowwolf/lede #1234: 研发请求: 在luci中实现 mux 参数设置

shelliao, 2019-04-27:
v2ray 中支持mux 设置，由于固件中默认设置为1，需要手工改动配置文件才能实现非默认设置
https://www.v2ray.com/chapter_02/mux.html
先请求把此参数放入luci中

xunqiuhuiyi, 2019-04-28:
实际使用感觉不明显，而且看官方解释 使用 Mux 看视频、下载或者测速通常都有反效果。

shelliao, 2019-04-29:
@xunqiuhuiyi  我看过文档，但是实际测试下来，针对不同的场景可以有不同的设置
例如有些朋友只是浏览网页的，我就希望这个值设到8，效果要比1要好，另外我试过在线路质量较差的场景下，是有一定效果的，特别是看网页。
如果另外一些朋友是看视频的，我才恢复成1
集成在luci上我就不需要帮人改底层文件了，直接让他们在界面上改即可

zxlhhyccc, 2019-05-02:
> @xunqiuhuiyi 我看过文档，但是实际测试下来，针对不同的场景可以有不同的设置
> 例如有些朋友只是浏览网页的，我就希望这个值设到8，效果要比1要好，另外我试过在线路质量较差的场景下，是有一定效果的，特别是看网页。
> 如果另外一些朋友是看视频的，我才恢复成1
> 集成在luci上我就不需要帮人改底层文件了，直接让他们在界面上改即可

luci界面上没有修改mux的地方：
[image]

shelliao, 2019-05-03:
@xunqiuhuiyi  Mux 是可以调数量的，请参考v2的官方文档，只是大侠没有在luci上做出来，我每次要ssh到固件中改比较麻烦. luci  Mux 设置只有enable/disalbe ， 数量是1

coolsnowwolf, 2019-05-04:
已经更新支持

closed 2019-08-29 by shelliao