coolsnowwolf/lede #1238: 我运行完  ./scripts/feeds update -a 以后再运行 ./scripts/feeds install -a 时候报错

jiushibushuo, 2019-04-28:
WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libpng', which does not exist
WARNING: Makefile 'package/lean/amule/Makefile' has a dependency on 'libupnp', which does not exist
WARNING: Makefile 'package/lean/amule/Makefile' has a build dependency on 'libgd', which does not exist
WARNING: Makefile 'package/lean/antileech/Makefile' has a build dependency on 'libgd', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'bc', which does not exist
WARNING: Makefile 'package/lean/autocore/Makefile' has a dependency on 'lm-sensors', which does not exist
WARNING: Makefile 'package/lean/automount/Makefile' has a dependency on 'kmod-fs-exfat', which does not exist
WARNING: Makefile 'package/lean/automount/Makefile' has a dependency on 'ntfs-3g', which does not exist
WARNING: Makefile 'package/lean/autosamba/Makefile' has a dependency on 'luci-app-samba', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libgnutls', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libopenldap', which does not exist
WARNING: Makefile 'package/network/utils/curl/Makefile' has a dependency on 'libssh2', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_aliyun/Makefile' has a dependency on 'ddns-scripts', which does not exist
WARNING: Makefile 'package/lean/ddns-scripts_aliyun/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/dnsforwarder/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/dnsforwarder/Makefile' has a dependency on 'coreutils-base64', which does not exist
WARNING: Makefile 'package/lean/frpc/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/ipv6-helper/Makefile' has a dependency on 'luci-proto-ipv6', which does not exist
WARNING: Makefile 'package/lean/ipv6-helper/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/kcptun/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/boot/kexec-tools/Makefile' has a dependency on 'liblzma', which does not exist
WARNING: Makefile 'package/lean/libidn2/Makefile' has a dependency on 'libunistring', which does not exist
WARNING: Makefile 'package/network/services/lldpd/Makefile' has a dependency on 'libnetsnmp', which does not exist
WARNING: Makefile 'package/lean/luci-app-accesscontrol/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-adbyby-plus/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/luci-app-adbyby-plus/Makefile' has a dependency on 'coreutils', which does not exist
WARNING: Makefile 'package/lean/luci-app-adbyby-plus/Makefile' has a dependency on 'coreutils-nohup', which does not exist
WARNING: Makefile 'package/lean/luci-app-adbyby-plus/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-amule/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-arpbind/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-autoreboot/Makefile' has a dependency on 'luci', which does not exist
WARNING: Makefile 'package/lean/luci-app-autoreboot/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-dnsforwarder/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-filetransfer/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-flowoffload/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-frpc/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/luci-app-frpc/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-guest-wifi/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-haproxy-tcp/Makefile' has a dependency on 'haproxy', which does not exist
WARNING: Makefile 'package/lean/luci-app-ipsec-vpnd/Makefile' has a dependency on 'strongswan-minimal', which does not exist
WARNING: Makefile 'package/lean/luci-app-ipsec-vpnd/Makefile' has a dependency on 'strongswan-mod-xauth-generic', which does not exist
WARNING: Makefile 'package/lean/luci-app-ipsec-vpnd/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-mwan3helper/Makefile' has a dependency on 'luci-app-mwan3', which does not exist
WARNING: Makefile 'package/lean/luci-app-mwan3helper/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-n2n_v2/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-nft-qos/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-openvpn-server/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-pppoe-relay/Makefile' has a dependency on 'rp-pppoe-common', which does not exist
WARNING: Makefile 'package/lean/luci-app-pppoe-relay/Makefile' has a dependency on 'rp-pppoe-relay', which does not exist
WARNING: Makefile 'package/lean/luci-app-pppoe-relay/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-pptp-server/Makefile' has a dependency on 'pptpd', which does not exist
WARNING: Makefile 'package/lean/luci-app-pptp-server/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-ramfree/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-samba4/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-sfe/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-ssr-plus/Makefile' has a dependency on 'coreutils', which does not exist
WARNING: Makefile 'package/lean/luci-app-ssr-plus/Makefile' has a dependency on 'coreutils-base64', which does not exist
WARNING: Makefile 'package/lean/luci-app-ssr-plus/Makefile' has a dependency on 'bash', which does not exist
WARNING: Makefile 'package/lean/luci-app-ssr-plus/Makefile' has a dependency on 'wget', which does not exist
WARNING: Makefile 'package/lean/luci-app-ssr-plus/Makefile' has a dependency on 'shadowsocks-libev-ss-redir', which does not exist
WARNING: Makefile 'package/lean/luci-app-ssrserver-python/Makefile' has a dependency on 'python', which does not exist
WARNING: Makefile 'package/lean/luci-app-ssrserver-python/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-syncdial/Makefile' has a dependency on 'luci-app-mwan3', which does not exist
WARNING: Makefile 'package/lean/luci-app-syncdial/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-ttyd/Makefile' has a dependency on 'ttyd', which does not exist
WARNING: Makefile 'package/lean/luci-app-ttyd/Makefile' has a build dependency on 'luci-base/host', which does not exist
WARNING: Makefile 'package/lean/luci-app-usb-printer/Makefile' has a dependency on 'p910nd', which does not exist
WARNING: Makefile 'package/lean/luci-app-usb-printer/Makefile' has a build dependency on 'luci-base/host', which does not exist
[... 14 lines cut ...]
WARNING: Makefile 'package/lean/open-vm-tools/Makefile' has a dependency on 'libtirpc', which does not exist
WARNING: Makefile 'package/lean/samba4/Makefile' has a dependency on 'libtirpc', which does not exist
[... cut ...]
WARNING: Makefile 'package/lean/shadowsocksr-libev/Makefile' has a dependency on 'libpcre', which does not exist
WARNING: Makefile 'package/lean/shadowsocksr-libev/Makefile' has a dependency on 'libpcre', which does not exist
WARNING: Makefile 'package/lean/shadowsocksr-libev/Makefile' has a dependency on 'libpcre', which does not exist
WARNING: Makefile 'package/lean/v2ray/Makefile' has a build dependency on 'golang/host', which does not exist
WARNING: Makefile 'package/lean/wxbase/Makefile' has a dependency on 'libexpat', which does not exist

wangxin6, 2019-04-28:
git pull

jiushibushuo, 2019-04-28:
> git pull

能说详细点吗？clone完怎么pull？

jiushibushuo, 2019-04-28:
> git pull

git pull 直接提示已经是最更新的版本，只好又install 了一遍，就没有错误提示，不懂了

closed 2019-04-28 by jiushibushuo