coolsnowwolf/lede #1255: x86_64 kernel 4.19.36 默认配置编译 运行时报错

KFERMercer, 2019-05-02:
内核日志：
`
[  131.835562] print_req_error: I/O error, dev loop0, sector 42792
[  131.841658] print_req_error: I/O error, dev loop0, sector 42824
[  131.847796] print_req_error: I/O error, dev loop0, sector 42856
[  131.853924] print_req_error: I/O error, dev loop0, sector 42896
[  131.860053] print_req_error: I/O error, dev loop0, sector 42912
[  131.866183] print_req_error: I/O error, dev loop0, sector 42968
[  131.872321] print_req_error: I/O error, dev loop0, sector 42992
[  131.878467] print_req_error: I/O error, dev loop0, sector 53832
[  131.884629] F2FS-fs (loop0): Issue discard(5349, 5349, 1) failed, ret: -5
[  131.891633] F2FS-fs (loop0): Issue discard(5353, 5353, 2) failed, ret: -5
[  131.898665] F2FS-fs (loop0): Issue discard(5357, 5357, 1) failed, ret: -5
[  131.905704] F2FS-fs (loop0): Issue discard(5362, 5362, 1) failed, ret: -5
[  131.912743] F2FS-fs (loop0): Issue discard(5364, 5364, 1) failed, ret: -5
[  131.919785] F2FS-fs (loop0): Issue discard(5371, 5371, 1) failed, ret: -5
[  131.926807] F2FS-fs (loop0): Issue discard(5374, 5374, 1) failed, ret: -5
[  131.933847] F2FS-fs (loop0): Issue discard(6729, 6729, 1) failed, ret: -5
`

编译选项摘要：
`
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
# CONFIG_TARGET_x86_generic is not set
# CONFIG_TARGET_x86_legacy is not set
# CONFIG_TARGET_x86_geode is not set
CONFIG_TARGET_x86_64_Generic=y
CONFIG_HAS_SUBTARGETS=y
CONFIG_TARGET_BOARD="x86"
CONFIG_TARGET_SUBTARGET="64"
CONFIG_TARGET_PROFILE="Generic"
CONFIG_TARGET_ARCH_PACKAGES="x86_64"
CONFIG_DEFAULT_TARGET_OPTIMIZATION="-Os -pipe"
CONFIG_CPU_TYPE=" "
CONFIG_LINUX_4_19=y
CONFIG_DEFAULT_alsa-utils=y
CONFIG_DEFAULT_amd64-microcode=y
CONFIG_EFI_IMAGES=y

`

而后在编译时取消勾选`libf2fs`和`mkf2fs`后，短时间没有看到此错误，但是路由一旦重启便会丢失所有配置。

coolsnowwolf, 2019-05-02:
f2fs 的 4.19 内核兼容性问题，已解决

KFERMercer, 2019-05-20:
5-20 4.19内核问题再次出现

closed 2019-05-20 by KFERMercer

liyuhaolol, 2019-08-07:
老大，请问你解决这个问题了么？我发现koolshare lede 2.32用的4.19.62核心依然有这个问题，满屏幕报错

KFERMercer, 2019-08-07:
> 老大，请问你解决这个问题了么？我发现koolshare lede 2.32用的4.19.62核心依然有这个问题，满屏幕报错

分区过大. rootfs设定不宜超过128m

zhiwen1987, 2020-05-28:
operation not supported error dev loop0 sector

内核 4.19.122
求解决方法