coolsnowwolf/lede #1258: mt_wifi编译报错

colinhd8, 2019-05-03:
拉取最新的代码(commit id:22e633a8d3eacbf80f788e979c116c3f79580e80)后编译K2P固件时会报错，具体信息如下：
os: Ubuntu 16.04.6 LTS
error info:
`el_24kc_musl/linux-ramips_mt7621/mt_wifi/mt_wifi_ap/../mt_wifi/embedded/wapp/wapp.o /home/colin/softwares/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt_wifi/mt_wifi_ap/../mt_wifi/embedded/
wapp/wapp.c
/home/colin/softwares/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt_wifi/mt_wifi_ap/../mt_wifi/embedded/wapp/wapp.c: In function 'wapp_send_apcli_query_rsp':
/home/colin/softwares/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt_wifi/mt_wifi_ap/../mt_wifi/embedded/wapp/wapp.c:401:28: error: 'AP_ADMIN_CONFIG {aka struct _AP_ADMIN_CONFIG}' has no me
mber named 'ApCliTab'
   apcli_entry = &pAd->ApCfg.ApCliTab[i];
                            ^
/home/colin/softwares/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt_wifi/mt_wifi_ap/../mt_wifi/embedded/wapp/wapp.c:403:17: error: 'AP_ADMIN_CONFIG {aka struct _AP_ADMIN_CONFIG}' has no me
mber named 'ApCliTab'
   if (pAd->ApCfg.ApCliTab[i].Valid == TRUE
                 ^
scripts/Makefile.build:326: recipe for target '/home/colin/softwares/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt_wifi/mt_wifi_ap/../mt_wifi/embedded/wapp/wapp.o' failed
make[5]: *** [/home/colin/softwares/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt_wifi/mt_wifi_ap/../mt_wifi/embedded/wapp/wapp.o] Error 1
Makefile:1532: recipe for target '_module_/home/colin/softwares/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt_wifi/mt_wifi_ap' failed
make[4]: *** [_module_/home/colin/softwares/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt_wifi/mt_wifi_ap] Error 2
make[4]: Leaving directory '/home/colin/softwares/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-4.14.114'
Makefile:228: recipe for target '/home/colin/softwares/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt_wifi/.built' failed
make[3]: *** [/home/colin/softwares/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/mt_wifi/.built] Error 2
make[3]: Leaving directory '/home/colin/softwares/openwrt/package/mtk/drivers/mt_wifi'
time: package/mtk/drivers/mt_wifi/compile#19.09#1.71#21.54
package/Makefile:107: recipe for target 'package/mtk/drivers/mt_wifi/compile' failed
make[2]: *** [package/mtk/drivers/mt_wifi/compile] Error 2
make[2]: Leaving directory '/home/colin/softwares/openwrt'
package/Makefile:103: recipe for target '/home/colin/softwares/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/colin/softwares/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/colin/softwares/openwrt'
/home/colin/softwares/openwrt/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2`
.config文件：
`#
# Automatically generated file; DO NOT EDIT.
# OpenWrt Configuration
#
CONFIG_MODULES=y
CONFIG_HAVE_DOT_CONFIG=y
# CONFIG_TARGET_sunxi is not set
# CONFIG_TARGET_apm821xx is not set
# CONFIG_TARGET_ath25 is not set
# CONFIG_TARGET_ar71xx is not set
# CONFIG_TARGET_ath79 is not set
# CONFIG_TARGET_brcm2708 is not set
# CONFIG_TARGET_bcm53xx is not set
# CONFIG_TARGET_brcm47xx is not set
# CONFIG_TARGET_brcm63xx is not set
# CONFIG_TARGET_cns3xxx is not set
# CONFIG_TARGET_octeon is not set
# CONFIG_TARGET_gemini is not set
# CONFIG_TARGET_mpc85xx is not set
# CONFIG_TARGET_imx6 is not set
# CONFIG_TARGET_mxs is not set
# CONFIG_TARGET_adm8668 is not set
# CONFIG_TARGET_adm5120 is not set
# CONFIG_TARGET_xburst is not set
# CONFIG_TARGET_ixp4xx is not set
# CONFIG_TARGET_lantiq is not set
# CONFIG_TARGET_malta is not set
# CONFIG_TARGET_pistachio is not set
# CONFIG_TARGET_mvebu is not set
# CONFIG_TARGET_kirkwood is not set
# CONFIG_TARGET_mediatek is not set
CONFIG_TARGET_ramips=y
# CONFIG_TARGET_at91 is not set
# CONFIG_TARGET_rb532 is not set
# CONFIG_TARGET_mcs814x is not set
# CONFIG_TARGET_layerscape is not set
# CONFIG_TARGET_octeontx is not set
# CONFIG_TARGET_oxnas is not set
[... 4944 lines cut ...]
# CONFIG_PACKAGE_bc is not set
# CONFIG_PACKAGE_bluelog is not set
# CONFIG_PACKAGE_bluez-daemon is not set
# CONFIG_PACKAGE_bluez-examples is not set
# CONFIG_PACKAGE_bluez-utils is not set
# CONFIG_PACKAGE_bonniexx is not set
# CONFIG_PACKAGE_bsdiff is not set
# CONFIG_PACKAGE_bspatch is not set
# CONFIG_PACKAGE_cal is not set
# CONFIG_PACKAGE_canutils is not set
# CONFIG_PACKAGE_cmdpad is not set
# CONFIG_PACKAGE_coap-client is not set
# CONFIG_PACKAGE_collectd is not set
CONFIG_PACKAGE_coremark=y
CONFIG_PACKAGE_coreutils=y
CONFIG_PACKAGE_coreutils-base64=y
# CONFIG_PACKAGE_coreutils-basename is not set
# CONFIG_PACKAGE_coreutils-cat is not set
# CONFIG_PACKAGE_coreutils-chcon is not set
# CONFIG_PACKAGE_coreutils-chgrp is not set
# CONFIG_PACKAGE_coreutils-chmod is not set
# CONFIG_PACKAGE_coreutils-chown is not set
# CONFIG_PACKAGE_coreutils-chroot is not set
# CONFIG_PACKAGE_coreutils-cksum is not set
# CONFIG_PACKAGE_coreutils-comm is

daofeng2015, 2019-05-03:
大屌好像把这个那个适配无线的删了,现在最新代码好像没有了

closed 2019-05-03 by coolsnowwolf

colinhd8, 2019-05-04:
嗯，确实没有了。不知道是原来的代码有问题还是有其它的原因。