coolsnowwolf/lede #1259: ac1300uhp 无法刷编译出的ac58u/ac1300uhp 固件

barton0426, 2019-05-03:
编译过程选择ipq 40xx下的ac58u/ac1300uhp
编译出来的是ac58u的sysupgrade bin文件
执行sysupgrade -v -n ***后 提示如下错误
root@OpenWrt:/tmp/upload# sysupgrade -n -v openwrt-ipq40xx-asus_rt-ac58u-squashfs-sysupgrade.bin
Device asus,rt-ac1300uhp not supported by this image
Supported devices: asus,rt-ac58u
Image check 'fwtool_check_image' failed.

hcym, 2019-05-03:
不知道啦，58u是没事的，就是广告过滤没有昨天大雕发布的网件7800彻底，用官方编译58u时完全没有问题，不懂还有什么小细节不对

barton0426, 2019-05-03:
> 编译过程选择ipq 40xx下的ac58u/ac1300uhp
> 编译出来的是ac58u的sysupgrade bin文件
> 执行sysupgrade -v -n ***后 提示如下错误
> root@OpenWrt:/tmp/upload# sysupgrade -n -v openwrt-ipq40xx-asus_rt-ac58u-squashfs-sysupgrade.bin
> Device [...]

不知道是不是现在ac1300uhp已经默认被识别为ac58u 都用ac58u的opboot

coolsnowwolf, 2019-05-03:
这两个是一样的，内部都叫做 AC58U

barton0426, 2019-05-03:
我上次是从官方刷到r9.1.1的 当时58u和1300在config的里还是分开的 这次编译的r9.5.1 就报错提示我上面贴的那个代码了

barton0426, 2019-05-09:
> 这两个是一样的，内部都叫做 AC58U

我能否通过修改system里的hostname来解决这个识别问题 fwtool_check_image 这个函数我也没能找到 到底是检查什么信息实现的

barton0426, 2019-06-10:
sysupgrade 有个参数是 -F 
自行解决，关

closed 2019-06-10 by barton0426

GitHubDer, 2019-12-25:
你好，弱弱的小白想问下你的1300UHP是怎么刷入opboot的？是像58U那样刷就可以了吗？