coolsnowwolf/lede #1260: 编译包含SoftEtherVPN的固件时报错镜像无效且无法访问

CarterJimmy, 2019-05-03:
make menuconfig时选中了SoftEtherVPN Bridge，Server以及client。然而编译过程中报错404。
  随后重新编辑menuconfig，只选中SoftEther Server并预先下载源码（`make download V=s`),然而还是报错404。复制链接至浏览器后确认为无效链接。
  希望可以更新一下相关地址，或者给出解决方案谢谢。
[image]
[image]
[image]

xinxijishuwyq, 2019-05-05:
https://github.com/Andy2244/openwrt-extra/tree/master/softethervpn5
上游更新过了，坐等大雕跟进

CarterJimmy, 2019-05-05:
@xinxijishuwyq 大雕的R9.1.5跟进了没？

xinxijishuwyq, 2019-05-07:
> @xinxijishuwyq 大雕的R9.1.5跟进了没？

没有，自己从那边撸过来吧。。
我不知道怎么给这边提交😂

fronttang, 2019-05-07:
> @xinxijishuwyq 大雕的R9.1.5跟进了没？

好像这个也编译失败。

CarterJimmy, 2019-05-08:
> > @xinxijishuwyq 大雕的R9.1.5跟进了没？
> 
> 好像这个也编译失败。

除了SoftEtherVPN编译报错外，noddos好像也会报错，取消选中后不知道是不是因为还是用依赖库残留的缘故，还是报错

fronttang, 2019-05-08:
> > > @xinxijishuwyq 大雕的R9.1.5跟进了没？
> > 
> > 
> > 好像这个也编译失败。
> 
> 除了SoftEtherVPN编译报错外，noddos好像也会报错，取消选中后不知道是不是因为还是用依赖库残留的缘故，还是报错

删除lean 下面的 softethervpn5 ,用上https://github.com/Andy2244/openwrt-extra/tree/master/softethervpn5 的，编译成功了。

CarterJimmy, 2019-05-16:
> > > > @xinxijishuwyq 大雕的R9.1.5跟进了没？
> > > 
> > > 
> > > 好像这个也编译失败。
> > 
> > 
> > 除了SoftEtherVPN编译报错外，noddos好像也会报错，取消选中后不知道是不是因为还是用依赖库残留的缘故，还是报错
> 
> 删除lean 下面的 softethervpn5 ,用上https://github.com/Andy2244/openwrt-extra/tree/master/softethervpn5 的，编译成功了。

谢谢，稍后尝试一下。
成功就关Issue

CarterJimmy, 2019-05-19:
> > > > @xinxijishuwyq 大雕的R9.1.5跟进了没？
> > > 
> > > 
> > > 好像这个也编译失败。
> > 
> > 
> > 除了SoftEtherVPN编译报错外，noddos好像也会报错，取消选中后不知道是不是因为还是用依赖库残留的缘故，还是报错
> 
> 删除lean 下面的 softethervpn5 ,用上https://github.com/Andy2244/openwrt-extra/tree/master/softethervpn5 的，编译成功了。

[image]
还是不行啊

closed 2019-09-29 by coolsnowwolf