coolsnowwolf/lede #1262: 树莓派3B+更新内核4.14.114后编译错误

jliveusa, 2019-05-03:
CC      drivers/mmc/host/bcm2835-mmc.o
In file included from ./include/linux/kernel.h:14:0,
                 from ./include/linux/delay.h:22,
                 from drivers/mmc/host/bcm2835-mmc.c:25:
drivers/mmc/host/bcm2835-mmc.c: In function 'bcm2835_mmc_dumpregs':
drivers/mmc/host/bcm2835-mmc.c:254:27: error: 'SDHCI_ACMD12_ERR' undeclared (first use in this function); did you mean 'SDHCI_CMD_CRC'?
   bcm2835_mmc_readw(host, SDHCI_ACMD12_ERR),
                           ^
./include/linux/printk.h:137:18: note: in definition of macro 'no_printk'
    printk(fmt, ##__VA_ARGS__); \
                  ^~~~~~~~~~~
drivers/mmc/host/bcm2835-mmc.c:253:2: note: in expansion of macro 'pr_debug'
  pr_debug(DRIVER_NAME ": AC12 err: 0x%08x | Slot int: 0x%08x\n",
  ^~~~~~~~
drivers/mmc/host/bcm2835-mmc.c:254:27: note: each undeclared identifier is reported only once for each function it appears in
   bcm2835_mmc_readw(host, SDHCI_ACMD12_ERR),
                           ^
./include/linux/printk.h:137:18: note: in definition of macro 'no_printk'
    printk(fmt, ##__VA_ARGS__); \
                  ^~~~~~~~~~~
drivers/mmc/host/bcm2835-mmc.c:253:2: note: in expansion of macro 'pr_debug'
  pr_debug(DRIVER_NAME ": AC12 err: 0x%08x | Slot int: 0x%08x\n",
  ^~~~~~~~
scripts/Makefile.build:326: recipe for target 'drivers/mmc/host/bcm2835-mmc.o' failed

coolsnowwolf, 2019-05-03:
上游的内核头文件改了

可以尝试git pull 后切换回 4.9 内核编译看看

hcym, 2019-05-04:
啥时候有n1撒，毕竟方便

coolsnowwolf, 2019-05-04:
已解决

closed 2019-05-05 by jliveusa