coolsnowwolf/lede #1264: 求助：挂载移动硬盘后不识别中文文件名

tsasan, 2019-05-04:
网件7500V2路由器，用lean大大的源码编译后，挂载移动硬盘成功，但所有中文文件夹和文件都消失了，只剩英文文件名的。怎么破，请指点。谢谢。

jackytang, 2019-05-04:
我使用ext4格式，没啥问题

tsasan, 2019-05-04:
我用的NTFS格式。以前用lean大大的老版本是可以识别的。肯定是有设置错了的地方。不知是哪里。烦恼中。

tsasan, 2019-05-07:
补充坏境说明：

硬件是 NETGEAR 7500V2 512M
编译环境是 THINKPAD T61 + ubuntu 14.06 32bit
源码是 lean 大的 lede 分支。

我在网上搜了很久，没有找到解决方案。此处高手如云，请各路英雄出手指点。谢谢。

jxph1123, 2019-05-13:
设置自动挂载中文
make kernel_menuconfig
File systems  --->
Native Language Support  --->
(utf8) Default NLS Option           ///此括号内为根据自己需要修改 必须改,网络共享默认中文
<*>   Simplified Chinese charset (CP936, GB2312)   可以不用改,默认
<*>   NLS ISO 8859-1  (Latin 1; Western European Languages)   可以不用改,默认
<*>   NLS UTF-8   可以不用改,默认
       DOS/FAT/NT Filesystems  --->
              (437) Default codepage for FAT   ///此括号内为根据自己需要修改
              (utf8) Default iocharset for FAT  ///此括号内为根据自己需要修改  必须改,USB默认中文

tsasan, 2019-05-26:
> 设置自动挂载中文
> make kernel_menuconfig
> File systems --->
> Native Language Support --->
> (utf8) Default NLS Option ///此括号内为根据自己需要修改 必须改,网络共享默认中文
> <_> Simplified Chinese charset (CP936, GB2312) 可以不用改,默认 <_> NLS [...]

谢谢指导。

closed 2019-05-26 by tsasan