coolsnowwolf/lede #1268: 希望将SSR plus里面的SSR服务器改成SS服务器

DavidDen, 2019-05-05:
感觉现在的ahead加密方式更适合于现在的网络环境，ssr服务器没有ahead加密方式，感谢开发者。

zcluo, 2019-05-05:
同意很少用ssr 

发自我的 iPhone

zxlhhyccc, 2019-05-05:
编译的时候勾选ss就是了

barton0426, 2019-05-05:
> 编译的时候勾选ss就是了

说的是ss server
非常支持大雕改成ss，libev版还简答

barton0426, 2019-05-05:
我每次都得手动安装 手动配置

DavidDen, 2019-05-05:
> 编译的时候勾选ss就是了
说明一下，我说的是ss server。谢谢Thx~

505575853, 2019-05-05:
https://github.com/505575853/lede/tree/R9/package/lean/luci-app-ssr-plus

我这个增加了

DavidDen, 2019-05-06:
> https://github.com/505575853/lede/tree/R9/package/lean/luci-app-ssr-plus
> 
> 我这个增加了

感谢你，真的，每次手动远程装，有点麻烦

505575853, 2019-05-06:
> > https://github.com/505575853/lede/tree/R9/package/lean/luci-app-ssr-plus
> > 我这个增加了
> 
> 感谢你，真的，每次手动远程装，有点麻烦

ssr ss只能同时启用一种别混着开

bigbighill, 2019-05-25:
我是直接用ss里的ss-redir 3.2.5替换了原版里的ss-redir 3.2.1 ,而且这个老版本和v2ray-plugin不兼容，顺道修改/etc/init.d/shadowsocksr 中的生成json文件的代码，把ktunp参数当作plugin_opts

https://github.com/shadowsocks/openwrt-shadowsocks/releases/tag/v3.2.5

jliveusa, 2019-07-24:
支持一下，请问各位大佬如何可否修改Makefile来更新SS版本，谢谢！

closed 2019-09-29 by coolsnowwolf