coolsnowwolf/lede #1272: 编译了三次都失败了，显示以下内容，请问是什么原因？

heroareslx, 2019-05-05:
Cleaning module download cache (golang/go#27455)
go: golang.org/x/net@v0.0.0-20180524181706-dfa909b99c79: unrecognized import path "golang.org/x/net" (https fetch: Get https://golang.org/x/net?go-get=1: dial tcp 216.239.37.1:443: i/o timeout)
go: golang.org/x/crypto@v0.0.0-20180505025534-4ec37c66abab: unrecognized import path "golang.org/x/crypto" (https fetch: Get https://golang.org/x/crypto?go-get=1: dial tcp 216.239.37.1:443: i/o timeout)
go: error loading module requirements

make[3]: *** [/home/heroares/lede/build_dir/target-x86_64_musl/frp-0.26.0/.built] Error 1
make[3]: Leaving directory `/home/heroares/lede/package/lean/frpc'
time: package/lean/frpc/compile#0.02#0.02#90.10
make[2]: *** [package/lean/frpc/compile] Error 2
make[2]: Leaving directory `/home/heroares/lede'
make[1]: *** [/home/heroares/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/heroares/lede'
make: *** [world] Error 2

zxlhhyccc, 2019-05-05:
需要在路由里设置，而且是SSR全局

Zxilly, 2019-05-06:
使用proxychain试试

tsasan, 2019-05-07:
我用的ssr-local+privoxy，实现ubuntu全局，成功编译。但是各项功能尚未完全成功。

jxph1123, 2019-05-14:
我也是卡在这里，怎么都不行，也翻了

whc2001, 2019-05-14:
似乎是这个页面失效了，浏览器打开会提示跳转
暂时解决方法是使用[goproxy](https://github.com/goproxyio/goproxy)，我这边可以成功
终端执行如下命令（可加入.bashrc）后再次尝试编译：
`export GOPROXY=https://goproxy.io`

peterwjp2017, 2019-05-29:
先make download
再make V=99

82kg, 2019-07-04:
> 似乎是这个页面失效了，浏览器打开会提示跳转
> 暂时解决方法是使用[goproxy](https://github.com/goproxyio/goproxy)，我这边可以成功
> 终端执行如下命令（可加入.bashrc）后再次尝试编译：
> `export GOPROXY=https://goproxy.io`

有用，谢谢

closed 2019-09-29 by coolsnowwolf

will-cxydiary, 2020-07-30:
> 似乎是这个页面失效了，浏览器打开会提示跳转
> 暂时解决方法是使用[goproxy](https://github.com/goproxyio/goproxy)，我这边可以成功
> 终端执行如下命令（可加入.bashrc）后再次尝试编译：
> `export GOPROXY=https://goproxy.io`

有用谢谢