coolsnowwolf/lede #1279: kernel升级到4.19.37后USB口丢了

tsasan, 2019-05-07:
如题：用4.14.114编译的，可以自动挂载移动硬盘，只是中文文件名丢失，只显示英文文件名。
pull新版本后，升级到4.19.37内核，编译后插上移动硬盘，连硬盘灯都不亮了。
还好我看中的酸酸乳还能正常喝。

求大神们出手指点。谢谢！

tsasan, 2019-05-07:
硬件是 NETGEAR 7500V2 512M
编译环境是 ubuntu 14.06 32bit 
源码是 lean 大的 lede 分支。

kob, 2019-05-09:
你得手动把USB支持的相关选项选上

tsasan, 2019-05-10:
选上了，base里的，kernel里的，util里的，包括automount,autosamba的脚本，都选上了。

后来我又用以前编译成功可以成功挂载移动硬盘的生成的映像文件夹中的config.in的配置，作为.config，重新编译新内核，编译过程还是成功的，就是USB硬盘挂上后连电源灯（硬盘的指示灯）都不会亮了。

以前版本的内核编译成功后，挂载移动硬盘灯（硬盘指示灯）亮，挂载也没问题，共享也没问题，就是中文文件名都不显示。

再以前从网上下载过一个更老版本的编译好的固件，里面没有zerotier，但是各项功能都正常。这次就是为了加上zerotier。登录到路由器上安装zerotier提示内核版本太低，不支持，这才重新搭环境重新编译的。结果一直没有完全成功。

yonggedebaqi, 2019-05-12:
我自己编译的完全没有问题，挂载硬盘中文显示正常。SSH登录支持显示中文输入。
[image]

yonggedebaqi, 2019-05-12:
硬盘中文都显示正常
[image]

yonggedebaqi, 2019-05-12:
wn系统访问路由器移动硬盘中文也是正常的。
[image]

myeveryheart, 2019-05-13:
@kob 请教，USB支持的相关选项选上，具体是哪些？

tsasan, 2019-05-16:
> 你得手动把USB支持的相关选项选上
请问kob兄，应该选哪些？我选的可能不对。

tsasan, 2019-05-16:
> 硬盘中文都显示正常
> [image]

天啊，请问怎么做到的？

kob, 2019-05-21:
参考这，打开usb支持https://blog.csdn.net/wynter_/article/details/78949502

tsasan, 2019-05-25:
> 参考这，打开usb支持https://blog.csdn.net/wynter_/article/details/78949502

谢谢！我去试试看。

tsasan, 2019-05-26:
终于成功了。谢谢上面大神的指导。原来错在没做 make kernel_menuconfig配置，另外支持中文还需那个 40-mount 脚本。

closed 2019-05-26 by tsasan