coolsnowwolf/lede #1281: 最新的源码编译没有v2ray

boer0327, 2019-05-08:
我拿了最新的版本编译了几次, 但就是没有v2ray, 我看了bin里面是有v2ray的, 但是就没有v2ray的luci. 之前的版本一直没有这个问题, 然后我发现之前那个单独的v2ray的选项已经在luci->application里面消失了. 

开始我以为是我本地有问题, 我还重新下载了一次, 发现还是这个问题. 大家有没有碰到这个问题.

zxlhhyccc, 2019-05-08:
用ssr-plus吧，大雕说原来的luci已经过时了

ziqqqq, 2019-05-08:
为啥我编译完成以后却没有ssr plus？

boer0327, 2019-05-08:
这个怎么选? 默认好像也没有选中吧?

boer0327, 2019-05-08:
@ziqqqq 我的也没有啊. 我自己找到包安装也没有. 这就尴尬了. 看样子不是我一个人的问题.

boer0327, 2019-05-08:
@zxlhhyccc 和上面的一个哥们一样, ssr-plus也没有出现. 手动从包里找到了安, 好像也没有反应.

boer0327, 2019-05-08:
@coolsnowwolf 大雕, 我找到问题了, 你的源码里面. ssr-plus luci的controller文件是乱码. 你可能需要重传一下. 我直接拿你前一个版本的文件替换我本地是没有问题的. 

@ziqqqq 朋友, 如果你不等着用, 可以等大雕更新了源码再用. 如果急用, 可以去拿上一个版本.

CarterJimmy, 2019-05-08:
> @zxlhhyccc 和上面的一个哥们一样, ssr-plus也没有出现. 手动从包里找到了安, 好像也没有反应.

安装后还需要输入代码启用的.
确保开启SSH并且Dropbear允许SSH访问的情况下进SSH后台输入 'echo 0xDEADBEFF > /etc/config/google_fu_mode' 即可

lifetyper, 2019-05-08:
提issue前请善用搜索
https://github.com/coolsnowwolf/lede/issues/881

boer0327, 2019-05-09:
@lifetyper @CarterJimmy 之前写这个, 还以为是v2ray. 哪知道最后找了一下是v2ray没有了. 看了你们的回复才知道是加密了.
@coolsnowwolf 不好意思了, 大雕, 原来是我搞错了. 

我现在关了.

closed 2019-05-09 by boer0327

jzhou404, 2020-05-22:
这个是固件编译好之后， 打开的开关吧？ echo 0xDEADBEEF> / etc / config / google_fu_mode

我也遇到问题， 是在make menuconfig的luci里， 根本就没有选择了啊？

请问， 加密了， 解密方法在哪里啊？