coolsnowwolf/lede #1284: 最新的K3的FRP不能用

kob, 2019-05-09:
运行显示，

root@tsk3:~# frpc
Illegal instruction

望修正

barton0426, 2019-05-09:
> 运行显示，
> 
> root@tsk3:~# frpc
> Illegal instruction
> 
> 望修正

试试frpc -h？

coolsnowwolf, 2019-05-12:
这个问题的根源是 K3 采用了 ARMv7 却删除了 FPU 浮点处理器的落后架构造成的

zxlhhyccc, 2019-05-14:
> 这个问题的根源是 K3 采用了 ARMv7 却删除了 FPU 浮点处理器的落后架构造成的

意思就是和v2ray一样upx压缩就无法运行对么？

coolsnowwolf, 2019-05-15:
不是压缩的问题
K3 的架构标称是 ARMv7 ，但标准 ARMv7 在 Golang 里面默认是带 FPU 的

实际上 Golang 里面定义 K3 应该是 ARMv6 

所以如果把 ARMv7 传递给 Golang 编译器，编译出来的 BIN 在 K3 上面实际不可用

wclmgcd, 2019-06-18:
> 
> 
> 不是压缩的问题
> K3 的架构标称是 ARMv7 ，但标准 ARMv7 在 Golang 里面默认是带 FPU 的
> 
> 实际上 Golang 里面定义 K3 应该是 ARMv6
> 
> 所以如果把 ARMv7 传递给 Golang 编译器，编译出来的 BIN 在 K3 上面实际不可用

就算编译时开启FPU,也不行?

wclmgcd, 2019-06-18:
大佬,用adbyby去广告效果不好,想在路由上安装AdGuardHome,K3出现Illegal instruction,看了这个问题才明白,大佬能否加上AdGuardHome用来去广告

yunnan0317, 2019-06-26:
released的arm版本可以直接用

yunnan0317, 2019-06-26:
直接下released for arm版本可以用

wclmgcd, 2019-06-26:
三个arm版本的都试过不行，你的是K3吗？

wclmgcd, 2019-06-26:
> 
> 
> 直接下released for arm版本可以用

路由器的安装教程好少，你是如何安装的

yunnan0317, 2019-06-27:
我是先用编译的，报illegal instruction，然后直接用released for arm替换/usr/bin/frpc就好了

closed 2019-10-18 by coolsnowwolf