coolsnowwolf/lede #1285: ssr的故障节点切换不够智能

myeveryheart, 2019-05-09:
比如，我的默认主服务器挂了（可ping通，但谷歌不可上）。
ssr会自动的切换到备用服务器。
当下一次检查的时候，它又检查到主服务器可用（但其实只是可ping通）。
于是它又自动切换到主服务器。
然后下个检查又检查到主服务器不可用，又切换到备用。
如此一直循环切。
感觉只有当应用了主服务器它才能正确理解“服务器挂了”。
其他情况只要可ping通它都会认为是可用的。

所以是否可以优化一下这里。

coolsnowwolf, 2019-05-12:
并不是 ping

myeveryheart, 2019-05-13:
哦哦好的，是我猜测错了。但实际体验貌似还能提高，下次我把log贴出来哈。感谢大佬。

myeveryheart, 2019-05-13:
2019-05-13 10:30:59 ShadowsocksR server switch OK
2019-05-13 10:30:55 V2Ray 4.19 (Po) Custom Started!
2019-05-13 10:30:50 Another server is avilable, now switching server.
2019-05-13 10:30:49 Current server error, try to switch another server.
2019-05-13 10:20:48 ShadowsocksR No Problem.
2019-05-13 10:10:48 ShadowsocksR No Problem.
switch to default [hk-1.lanan.world] proxy!
2019-05-13 10:00:43 V2Ray 4.19 (Po) Custom Started!
2019-05-13 10:00:38 Main server is avilable.
2019-05-13 10:00:36 Current server is not default Main server, try to switch back.
2019-05-13 09:50:36 ShadowsocksR server switch OK
2019-05-13 09:50:31 V2Ray 4.19 (Po) Custom Started!
2019-05-13 09:50:27 Another server is avilable, now switching server.
2019-05-13 09:50:26 Current server error, try to switch another server.

myeveryheart, 2019-05-13:
大佬请看。主服务器是[hk-1.lanan.world]，然后被发现Current server error, try to switch another server.
过一会又检测Main server is avilable于是切回去了。

myeveryheart, 2019-05-13:
不过我发现一个诡异的事情，“状态”那里检查服务器，OK的未必真的OK，还得自己改主服务器，然后检查谷歌联通性才算真的测试。所以是不是可以把“检查服务器”和“故障切换”都以“谷歌连通性”作为判断依据呢。
[image]
使用的是v2ray

myeveryheart, 2019-05-13:
如上图，其中一半的OK其实都是不能上谷歌的。

closed 2019-10-18 by coolsnowwolf