coolsnowwolf/lede #1287: 报错 libopenssl libmbedtls冲突的问题

waybq, 2019-05-10:
`  < > libgnutls....................... GNU TLS (library) (NEW)
  -*- libmbedtls................. Embedded SSL (library) (NEW)
  -*- libopenssl.... Open source SSL toolkit (libraries)  --->
  < > libpolarssl................ Embedded SSL (library) (NEW)
  < > libwolfssl........................ wolfSSL library (NEW)`

当这两个都被选中的时候编译会报错
`Configuring fwtCollected errors:
 * check_data_file_clashes: Package libustream-openssl wants to install file /openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/lib/libustream-ssl.so
	But that file is already provided by package  * libustream-mbedtls
 * opkg_install_cmd: Cannot install package libustream-openssl.
 * check_data_file_clashes: Package openvpn-mbedtls wants to install file /openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/etc/init.d/openvpn
	But that file is already provided by package  * openvpn-openssl
 * check_data_file_clashes: Package openvpn-mbedtls wants to install file /openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/lib/upgrade/keep.d/openvpn
	But that file is already provided by package  * openvpn-openssl
 * check_data_file_clashes: Package openvpn-mbedtls wants to install file /openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/usr/sbin/openvpn
	But that file is already provided by package  * openvpn-openssl
 * check_data_file_clashes: Package openvpn-mbedtls wants to install file /openwrt/lede/build_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/usr/share/openvpn/openvpn.options
	But that file is already provided by package  * openvpn-openssl
 * opkg_install_cmd: Cannot install package openvpn-mbedtls.`

而在用https://github.com/openwrt/openwrt的源码的时候，这两个都存在也不会报错能编译成功。

SoPudge, 2019-05-12:
编译curl也是类似问题，需要libmbedtls，但是实际上这个包在编译ss的时候就已经安装了，好像是有冲突

SuLingGG, 2019-05-12:
也遇到了类似的问题，在最后安装软件包的时候报出以下错误：

```
Configuring brcmfmac-firmware-43430-sdio.
Configuring ppp-mod-pppoe.
Configuring luci-i18n-nlbwmon-zh-cn.
Configuring kmod-net-rtl8192su.
Collected errors:
 * check_data_file_clashes: Package libustream-openssl wants to install file /home/admin/workspace/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-brcm2708/lib/libustream-ssl.so
        But that file is already provided by package  * libustream-mbedtls
 * opkg_install_cmd: Cannot install package libustream-openssl.
make[2]: *** [package/install] Error 255
make[2]: Leaving directory `/home/admin/workspace/lede'
make[1]: *** [/home/admin/workspace/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_install] Error 2
make[1]: Leaving directory `/home/admin/workspace/lede'
make: *** [world] Error 2
```

直接删除 .config 文件，重新选择 CPU 平台和型号后，然后发现 Libraries - SSL - libopenssl 这个选项默认为 -*- 状态，不可取消选中。

coolsnowwolf, 2019-05-12:
不支持同一个系统存在两个 ssl 库，请自行解决

LGA1150, 2019-05-15:
冲突的不是这两个
是 libustream
二选一就行了

bigbighill, 2019-05-21:
也遇到这个问题，但两者默认同时选中了，无法取消选择libopenssl，也搞不清是哪个依赖这个包

closed 2019-10-18 by coolsnowwolf