coolsnowwolf/lede #1289: glibc编译报错perl.c:311:5: error: ‘_LIB_VERSION’ undeclared (first use in this function); did you mean ‘_SC_VERSION’?

colinhd8, 2019-05-10:
Advanced configuration options (for developers)
-- Toolchain Options
----C library implemetation 
选择glibc的话，编译时会报错：
perl.c:311:5: error: ‘_LIB_VERSION’ undeclared (first use in this function); did you mean ‘_SC_VERSION’?

看了一下官方的perl补丁，对此问题有作了修复：
packages/lang/perl/patches/010-musl-compat.patch
perl: fix compilation with glibc 2.27 2018/10/16
具体变更如下
-#if defined(LIBM_LIB_VERSION)
+#if defined(LIBM_LIB_VERSION) && defined(__UCLIBC__)

而大雕的源里还是这样的：
-#if defined(LIBM_LIB_VERSION)
+#if defined(LIBM_LIB_VERSION) && (defined(__GLIBC__) || defined(__UCLIBC__))
所以导致了报错。但看了下日期，是2018/10/18同步的。不知这个GLIBC是有意加上去的吗？
可否同步成官方一样的内容？

先行谢过。

zxlhhyccc, 2019-05-11:
你编译什么会报错？我用这个编译正常，但用uClibc会报错。。。。。

zxlhhyccc, 2019-05-11:
还有就是，你应该可以比照官方的加进去试试。。。。

colinhd8, 2019-05-11:
编译x86-64的固件会报错。我照着官方的改了，不会再报这个错了(但到后面步骤时报corecpumark的错)。
从代码上看,uclibc是会走到这个IF语句里面的。所以如果是x86-64的话，应该也是会报错。

coolsnowwolf, 2019-05-12:
不再兼容uclibc，请自行解决

colinhd8, 2019-05-12:
关键是现在glibc也会报错。

closed 2019-10-18 by coolsnowwolf