coolsnowwolf/lede #1291: 希望恢复由源码编译v2ray, 而不是直接放出二进制文件

KFERMercer, 2019-05-11:
如题

zxlhhyccc, 2019-05-11:
源码编译有瑕疵，如果源码编译k3时候，文件不能压缩，否则无法运行。。。。

coolsnowwolf, 2019-05-12:
这样说吧，不给二进制你们说 K3 不能运行，给了修复好的你们说要源代码的，要不你们先打一架?

coolsnowwolf, 2019-05-12:
这个问题的根源是 K3 采用了 ARMv7 却删除了 FPU 浮点处理器的落后架构造成的

colinhd8, 2019-05-12:
是否有办法把两个都带上，然后根据编译目标来决定是使用源码编译，还是直接使用二进制？

KFERMercer, 2019-05-12:
我个人认为应该针对个别target来提供二进制文件，而不是因为个别机型有问题而牺牲整体的透明度。

coolsnowwolf, 2019-05-12:
实际上我真的不想支持 K3 这种bcm家的东西了

wangxin6, 2019-05-13:
支持源码编译方式

barton0426, 2019-05-14:
你们至少得提一下源码编译的理由吧。。。大雕也是为了照顾k3的用户

colinhd8, 2019-05-14:
KFERMercer已经写得很清楚了吧。

KFERMercer, 2019-05-14:
还是希望重新放出v2ray源码。直接采用二进制会让人以为lean有像k论坛看齐的意思

zxlhhyccc, 2019-05-14:
> 还是希望重新放出v2ray源码。直接采用二进制会让人以为lean有向k论坛看齐的意思

你很自私。。。。。。

zxlhhyccc, 2019-05-14:
> 还是希望重新放出v2ray源码。直接采用二进制会让人以为lean有向k论坛看齐的意思

用二进制编译和用源码编译我不知道有什么区别，我认为只要大家都能使用就是最好的，支持大佬。。。

jsylin, 2019-05-26:
编译不一定大家都会安装 但二进制包大家却都会安装 你这么说那只有你能安装v2ray 其他人怎么办？能不要这么自私吗？

jsylin, 2019-05-26:
> 还是希望重新放出v2ray源码。直接采用二进制会让人以为lean有向k论坛看齐的意思

编译不一定大家都会安装 但二进制包大家却都会安装 你这么说那只有你能安装v2ray 其他人怎么办？能不要这么自私吗？

colinhd8, 2019-05-26:
楼上的言论我也是醉了，照你这么说直接放出编译好的整个openwrt的二进制文件不是更方便？既然来这里的人本来就是要拿源码来编码的，那么他会不知道怎么编译一个功能包吗？再说了，希望老大放出源码形式，那肯定要配套的东西整合好，大家只要勾选这个包即可，这样的话放出二进制还是源码对你编译来说有区别？

另外看到说LZ自私的，我怎么觉得他只是希望这个项目变得更好？有体现出不让别人用v2ray的意思吗？

KFERMercer, 2019-05-26:
> > 还是希望重新放出v2ray源码。直接采用二进制会让人以为lean有向k论坛看齐的意思
> 
> 编译不一定大家都会安装 但二进制包大家却都会安装 你这么说那只有你能安装v2ray 其他人怎么办？能不要这么自私吗？

不明白你想表达什么意思

closed 2019-10-18 by coolsnowwolf