coolsnowwolf/lede #1294: Phicomm K3 2.4G WIFI最高只支持72M

xuansongzhang, 2019-05-11:
最近发现K3的2.4G WIFI带宽最高只能支持到72M，相同的设备（百度影棒）连K3 2.4G显示带宽是72M，连华为WS5200带宽是150M。

K3 2.4G设置，模式n，信道11，频宽20M，功率20dBm。

BROBIRD, 2019-05-13:
> 802.11n支持在标准带宽（20MHz）上的速率包括有（单位Mbit/s）：7.2, 14.4, 21.7, 28.9, 43.3, 57.8, 65, 72.2（短保护间隔，单数据流）。
https://zh.wikipedia.org/zh-cn/IEEE_802.11n#%E6%8A%80%E8%A1%93%E7%89%B9%E9%BB%9E
自己不会先查查么？

zxlhhyccc, 2019-05-14:
> > 802.11n支持在标准带宽（20MHz）上的速率包括有（单位Mbit/s）：7.2, 14.4, 21.7, 28.9, 43.3, 57.8, 65, 72.2（短保护间隔，单数据流）。
> 
> https://zh.wikipedia.org/zh-cn/IEEE_802.11n#%E6%8A%80%E8%A1%93%E7%89%B9%E9%BB%9E
> 自己不会先查查么？

不过确实，即使设置默认和强制40频宽，路由还是默认在20频宽下。。。

zxlhhyccc, 2019-05-16:
> 最近发现K3的2.4G WIFI带宽最高只能支持到72M，相同的设备（百度影棒）连K3 2.4G显示带宽是72M，连华为WS5200带宽是150M。
> 
> K3 2.4G设置，模式n，信道11，频宽20M，功率20dBm。

经实际使用，单天线最高72.2，双天线最高144.4

xuansongzhang, 2019-05-16:
> 最近发现K3的2.4G WIFI带宽最高只能支持到72M，相同的设备（百度影棒）连K3 2.4G显示带宽是72M，连华为WS5200带宽是150M。
> K3 2.4G设置，模式n，信道11，频宽20M，功率20dBm。
> 
> 经实际使用，单天线最高72.2，双天线最高144.4

不知道百度影棒是几天线，反正连华为的WS5200后，百度影棒的网络状态里面显示150M，连K3后，百度影棒的网络状态里面显示72M。

closed 2019-05-16 by xuansongzhang

xiao-mantou, 2021-04-17:
固件或驱动问题,即使实际是40Mhz也会协商72Mbps,更换固件解决.