coolsnowwolf/lede #1295: 请教大神ssr-plus这个错误怎么解决

ziqqqq, 2019-05-12:
make[3]: Entering directory '/home/ll/lede/package/lean/luci-app-ssr-plus'
rm -rf /home/ll/lede/build_dir/target-mips_24kc_musl/luci-app-ssr-plus-1/.pkgdir/luci-app-ssr-plus.installed /home/ll/lede/build_dir/target-mips_24kc_musl/luci-app-ssr-plus-1/.pkgdir/luci-app-ssr-plus
mkdir -p /home/ll/lede/build_dir/target-mips_24kc_musl/luci-app-ssr-plus-1/.pkgdir/luci-app-ssr-plus
install -d -m0755 /home/ll/lede/build_dir/target-mips_24kc_musl/luci-app-ssr-plus-1/.pkgdir/luci-app-ssr-plus/usr/lib/lua/luci
cp -pR ./luasrc/* /home/ll/lede/build_dir/target-mips_24kc_musl/luci-app-ssr-plus-1/.pkgdir/luci-app-ssr-plus/usr/lib/lua/luci
cp: cannot open './luasrc/controller/shadowsocksr.lua' for reading: Permission denied
Makefile:86: recipe for target '/home/ll/lede/build_dir/target-mips_24kc_musl/luci-app-ssr-plus-1/.pkgdir/luci-app-ssr-plus.installed' failed
make[3]: *** [/home/ll/lede/build_dir/target-mips_24kc_musl/luci-app-ssr-plus-1/.pkgdir/luci-app-ssr-plus.installed] Error 1
make[3]: Leaving directory '/home/ll/lede/package/lean/luci-app-ssr-plus'
time: package/lean/luci-app-ssr-plus/compile#0.14#1.98#3.28
package/Makefile:107: recipe for target 'package/lean/luci-app-ssr-plus/compile' failed
make[2]: *** [package/lean/luci-app-ssr-plus/compile] Error 2
make[2]: Leaving directory '/home/ll/lede'
package/Makefile:103: recipe for target '/home/ll/lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/ll/lede/staging_dir/target-mips_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ll/lede'
/home/ll/lede/include/toplevel.mk:216: recipe for target 'world' failed
make: *** [world] Error 2

SasamiMasakiJurai, 2019-05-12:
cannot open './luasrc/controller/shadowsocksr.lua' for reading: Permission denied

你依赖什么有没有装全？？？？

相关依赖有没有编译完毕？？？？

ziqqqq, 2019-05-12:
之前替换过shadowsocksr.lua

BROBIRD, 2019-05-12:
`cp: cannot open './luasrc/controller/shadowsocksr.lua' for reading: Permission denied`
已经很明确告诉你是权限问题了

lifetyper, 2019-05-13:
不要用root操作文件，然后又用其他非root账号编译。

ziqqqq, 2019-05-13:
我也知道是权限问题，不是在root下，我问的是怎么解决这个问题？

lifetyper, 2019-05-13:
把这个文件删掉，用你自己的编译用户重新pull 代码替换掉这个文件。

closed 2019-05-15 by coolsnowwolf