coolsnowwolf/lede #1296: ssr dns解析等相关问题

colinhd8, 2019-05-12:
固件版本:  OpenWrt R9.5.1 By Lean / LuCI Master (git-19.121.62139-9ee402e) x86-64

问题1:SSR中DNS解析方式如果选择使用本机端口为5335的DNS服务，则在状态页检查连接性会报连接错误。
5335端口是pdnsd的端口，可选择这个选项后，倒是把pdnsd停掉了，所以5335端口就没了。而/etc/dnsmasq.ssr/gfw_list.conf中的规则却还是使用的5335端口。所以导致连接错误。
里面倒是还有一个dnscache的进程，一直开着，端口是5333，但貌似没人用，不知道起啥作用

问题2: SSR设置中只对v2ray开启了透明代理，没有可以设置SOCKS5代理的地方，是否考虑加一个选项？当然目前应该是可以手动改/var/etc/v2-ssr-retcp.json。
而目前设置中的SOCKS5代理只针对SSR服务。也就是说得先配置了SSR代理这个才有作用。

问题3: SSR配置了KCPTUN，但在状态页看有时候显示运行中，有时显示未运行，实际查看进程的话，也确实是有时候是有kcptun-client进程的，有时候又没了。感觉就是不稳定。另外，ssr-local貌似不走kcptun.

1779729350, 2019-05-12:
5353端口是要和chinadns或者别的应用综合使用的
其他的不用不了解

coolsnowwolf, 2019-05-12:
5353端口留给你自己安装其他的防污染dns 用的，例如 chinadns，dnscrypt 等

v2ray socks5 以后再说吧

kcptun 自从bbr 出来后，接近淘汰了

colinhd8, 2019-05-12:
问题1明白了。就是这个选项不提供默认的DNS实现，而是需要自己安装一个。
问题2那就等更新吧。
问题3可以用BBR代替。

谢谢解答。

closed 2019-05-12 by colinhd8

willisworley, 2019-07-14:
看来 l大有望 集成 pcap_dnsproxy 啊，for openwrt

YuefengLi, 2020-04-12:
我的 pandorabox 自带里面有一个 dns 加速的选项，点了之后就会启动 dnscache 在 5333