coolsnowwolf/lede #1298: tvheadend最新是4.3版，源码里还是4.0.10，望升级一下，谢谢！

zxlhhyccc, 2019-05-12:
[image]
[image]

zxlhhyccc, 2019-05-12:
[image]

yonggedebaqi, 2019-05-14:
自己改 HASH值256
版本号自己改一下

zxlhhyccc, 2019-05-14:
> 自己改 HASH值256
> 版本号自己改一下

改了，编译报错。。。。

yonggedebaqi, 2019-05-18:
我试着编译4.3 修改Makefile文集 注释掉PKG_MIRROR_HASH这行
添加
CONFIGURE_ARGS += 
--arch=$(ARCH) 
--disable-dbus_1 
--disable-libav 
--enable-bundle 
--disable-ffmpeg_static 
--disable-cccam 
--disable-libx264 
--disable-cwc 
--disable-capmt 
--disable-constcw
编译也是报错

SoPudge, 2019-05-19:
我没有试着编译这个包，但是我试过修改udpxy的版本，你可以试试注释掉hash这行，另外只修改version值为4.3，官方op源里有这个包，所以依赖什么的应当没啥问题，理论上只修改makefile就可以更新

closed 2019-10-18 by coolsnowwolf