coolsnowwolf/lede #1299: 昨天更新代码编译的v2ray，编辑和添加节点错误？

snakwu, 2019-05-13:
500 内部服务器错误
对不起，服务器遇到未知错误。

Failed to execute arcombine dispatcher target for entry '/admin/services/shadowsocksr/servers/cfg074a8f'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/map'.
A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/nsection'.
A runtime error occurred: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'cbi/ucisection'.
A runtime error occurred: /usr/lib/lua/luci/template.lua:74: Failed to load template 'cbi/other_dvalue'.
Error while parsing template '/usr/lib/lua/luci/view/cbi/other_dvalue.htm':
No such file or directory
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:74: in function '__init__'
	/usr/lib/lua/luci/util.lua:65: in function 'Template'
	/usr/lib/lua/luci/template.lua:27: in function 'render'
	/usr/lib/lua/luci/cbi.lua:257: in function 'render'
	/usr/lib/lua/luci/cbi.lua:1461: in function 'render'
	/usr/lib/lua/luci/cbi.lua:266: in function 'render_children'
	[string "/usr/lib/lua/luci/view/cbi/ucisection.htm"]:7: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 'render'
	/usr/lib/lua/luci/dispatcher.lua:359: in function 'include'
	[string "/usr/lib/lua/luci/view/cbi/nsection.htm"]:1: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 
	(tail call): ?
	/usr/lib/lua/luci/cbi.lua:257: in function 'render'
	/usr/lib/lua/luci/cbi.lua:266: in function 'render_children'
	[string "/usr/lib/lua/luci/view/cbi/map.htm"]:6: in main chunk
stack traceback:
	[C]: in function 'error'
	/usr/lib/lua/luci/template.lua:97: in function 
	(tail call): ?
	/usr/lib/lua/luci/cbi.lua:257: in function 'render'
	/usr/lib/lua/luci/cbi.lua:440: in function 'render'
	/usr/lib/lua/luci/dispatcher.lua:948: in function 'target'
	/usr/lib/lua/luci/dispatcher.lua:984: in function

coolsnowwolf, 2019-05-13:
空间满了

snakwu, 2019-05-13:
但我的可用空间是还有的，指的那个空间满了？

snakwu, 2019-05-13:
我检查了一下，空间并没有满呢

snakwu, 2019-05-13:
Filesystem                Size      Used Available Use% Mounted on
/dev/root                25.8M     25.8M         0 100% /rom
tmpfs                   998.8M      2.5M    996.3M   0% /tmp
/dev/loop0              996.3M     19.3M    841.1M   2% /overlay
overlayfs:/overlay      996.3M     19.3M    841.1M   2% /
/dev/sda1               252.0M      4.0M    242.9M   2% /boot
/dev/sda1               252.0M      4.0M    242.9M   2% /boot
tmpfs                   512.0K         0    512.0K   0% /dev
tmpfs                   998.8M      2.5M    996.3M   0% /usr/share/adbyby/data

505575853, 2019-05-13:
Error while parsing template '/usr/lib/lua/luci/view/cbi/other_dvalue.htm':
No such file or directory

coolsnowwolf, 2019-05-13:
No such file or directory 看着是文件系统丢失文件了

snakwu, 2019-05-13:
对的，但不知道怎么会少文件

closed 2019-10-18 by coolsnowwolf