coolsnowwolf/lede #1309: 编译K3，bin文件夹里生产的其他路由器的固件可以直接用来刷机吗？

xuansongzhang, 2019-05-16:
我用树莓派安装ubuntu64-rpi进行编译的，编译的过程中提示/usr/include/asm-generic/int-ll64.h中__u64的定义冲突与/home/ubuntu/lede/build_dir/host/u-boot-2018.03/include/compiler.h里的
typedef uint64_t __u64冲突，试图把所有/usr/include里面的关于__u64的定义全部改掉（改成__ull64）后编译，但是最终还是出现与__u64相关的错误。最终直接把/home/ubuntu/lede/build_dir/host/u-boot-2018.03/include/compiler.h里的typedef uint64_t __u64注释掉后编译成功了。
现在的问题是成功后bin文件夹里也产生了使用相同CPU的其他路由器的固件，请教一下这些其他路由器的固件能不能直接用来其对应路由器的刷机。

myeveryheart, 2019-05-16:
我也有相似的疑问。编译选择的K3，但是生成了其他bcm53的固件，是否这样会花费更多编译时间？

coolsnowwolf, 2019-05-16:
默认 op 官方就是这样，等待上游更新吧

xuansongzhang, 2019-05-16:
我

> 默认 op 官方就是这样，等待上游更新吧

谢谢！其他的应该不能用来刷机吧？

hcym, 2019-05-16:
论坛不是有一个大神的k3 gitpull的帖子吗，里面特地提到怎么不用编译其它的机型，至于能不能用应该问题不大，有问题无法信号大小

xuansongzhang, 2019-05-16:
作了一个实验：编译ARRIS SBR-AC3200的固件，用bin目录下的K3固件刷K3后无线没信号了。

closed 2019-05-16 by xuansongzhang

xuansongzhang, 2019-08-28:
@mzy22mzy我不会改CFE，也没想到这一招。OpenWrt的无线驱动太老，不想用OpenWrt了，ARRIS原厂固件也没备份，网上下载的img格式的直接在CFE下刷了也没用，现在回不去了。你有这个路由器吗？有的话备份一下原厂固件给我发一份。谢谢！zhangxuansong@hotmail.com

xuansongzhang, 2019-08-28:
@mzy2240 试过华硕RT-AC3200的梅林固件和原厂固件，完全刷不了，提示I/O error。error。Koolshare论坛上的RT-AC3200_380.70_0-X7.9.1版本能正常刷机，但是一直循环重启，无法开机。

mzy2240, 2019-08-28:
@xuansongzhang 我打算入手一个刷机练练手 如果我拿到了会发你邮件的 BTW 你还记得用Arris自带固件的时候热量大吗？刚摸了下TM-AC1900 温度也就30-40

xuansongzhang, 2019-08-28:
@mzy2240 用ARRIS原厂最新的1.07版固件发热量蛮大的，刷了OpenWrt后温度差不多，室温28度左右时，CPU温度接近100度了。温度太高了，不建议买这款。可以买K3刷机，还不用拆机就能用CFE miniweb刷机。

mzy2240, 2019-08-28:
@xuansongzhang 这温度太夸张了 看来还是买TMAC1900吧 本来search了一番发现它是理论上能刷支持AiMesh的华硕固件的路由器里最便宜的 看来这种电源内置方案还是有点问题

xuansongzhang, 2019-08-28:
@mzy2240 温度高可能是因为为了外观好看，散热设计不好的原因。ARRIS改CFE刷华硕原厂固件的方法给我学习一下。

mzy2240, 2019-08-29:
@xuansongzhang 没有现成的 我原本打算参照TM-AC1900, Tenda AC18 还有 EA6900 的教程来操作 我大概看了 流程差不多

xuansongzhang, 2019-08-29:
谢谢！我找找看。如果你最终还是决定买ARRIS SBR-AC3200P，帮忙备份原厂固件发我一份。

mzy2240, 2019-08-29:
@xuansongzhang 没问题👌

yaobohan888, 2019-10-22:
@mzy2240  arris 改cfe的方法，能指导一下吗？谢谢