coolsnowwolf/lede #1313: 希望大大,能为v2ray加进自动定dns服务器功能,和加进支持dnsforwarder方式转发dns,谢谢!

snakwu, 2019-05-16:
希望大大,能为v2ray加进自动定dns服务器功能而不是只能在指定的dns列表中选择,和加进支持dnsforwarder方式转发dns,谢谢!
因为pdns有缓存问题,dnsforwarder优于pdns,谢谢!

coolsnowwolf, 2019-05-16:
做了预留了，用本地的 5335 端口监听 dnsforwader

[image]

snakwu, 2019-05-17:
因为我是自己搭建的dns，所以希望加进可以自定义dns项，还有如果可以，也希望增加udp的方式来解释dns，谢谢你这么好用的项目。。。

505575853, 2019-05-17:
client.lua
-o = s:option(ListValue, "tunnel_forward", translate("Anti-pollution DNS Server"))
+o = s:option(Value, "tunnel_forward", translate("Anti-pollution DNS Server"))

snakwu, 2019-05-17:
> client.lua
> -o = s:option(ListValue, "tunnel_forward", translate("Anti-pollution DNS Server"))
> +o = s:option(Value, "tunnel_forward", translate("Anti-pollution DNS Server"))

谢谢,为什么你也把那部分东西给删除了?

snakwu, 2019-05-17:
> client.lua
> -o = s:option(ListValue, "tunnel_forward", translate("Anti-pollution DNS Server"))
> +o = s:option(Value, "tunnel_forward", translate("Anti-pollution DNS Server"))

我的email:snakwu@hotmail.com,能联系一下吗?谢谢!

closed 2019-10-18 by coolsnowwolf