coolsnowwolf/lede #1337: 编译时 fullconenat-2018-12-15报错，怎么解决。

qinfeng8848, 2019-05-21:
touch /home/qf/lede/staging_dir/target-mipsel_24kc_musl/root-ramips/stamp/.libxtables_installed
echo "libxtables" >> /home/qf/lede/staging_dir/target-mipsel_24kc_musl/pkginfo/iptables.default.install
make[3]: Leaving directory `/home/qf/lede/package/network/utils/iptables'
time: package/network/utils/iptables/compile#28.73#3.52#38.64
make[3]: Entering directory `/home/qf/lede/package/lean/openwrt-fullconenat'
touch /home/qf/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/fullconenat-2018-12-15-d4daedd0/.prepared_5b13fd5a892ddd2dfe57c5400e3f7e63_6664517399ebbbc92a37c5bb081b5c53_check
. /home/qf/lede/include/shell.sh; xzcat /home/qf/lede/dl/fullconenat-2018-12-15-d4daedd0.tar.xz | tar -C /home/qf/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/fullconenat-2018-12-15-d4daedd0/.. -xf -
tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors
make[3]: *** [/home/qf/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/fullconenat-2018-12-15-d4daedd0/.prepared_5b13fd5a892ddd2dfe57c5400e3f7e63_6664517399ebbbc92a37c5bb081b5c53] Error 2
make[3]: Leaving directory `/home/qf/lede/package/lean/openwrt-fullconenat'
time: package/lean/openwrt-fullconenat/compile#0.10#0.08#0.72
make[2]: *** [package/lean/openwrt-fullconenat/compile] Error 2
make[2]: Leaving directory `/home/qf/lede'
make[1]: *** [/home/qf/lede/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory `/home/qf/lede'
make: *** [world] Error 2

coolsnowwolf, 2019-05-22:
被墙了，删除 对应的 tar 文件重新下载

closed 2019-05-27 by coolsnowwolf

lifeaAngle, 2020-01-06:
请问你怎么解决的