coolsnowwolf/lede #1342: 重启路由器，ssr plus总是未运行状态

bigbighill, 2019-05-26:
重启路由器，ssr plus总是未运行状态，要手动启动？默认如此还是bug？

bigbighill, 2019-05-26:
/etc/rc.d/S60flowoffload:  * Running script '/var/etc/shadowsocksr.include'
Sun May 26 11:42:09 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:    ! Skipping due to path error: No such file or directory

bigbighill, 2019-05-26:
```
Sun May 26 16:16:19 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:  * Running script '/var/etc/shadowsocksr.include'
Sun May 26 16:16:19 2019 daemon.notice procd: /etc/rc.d/S60flowoffload:    ! Skipping due to path error: No such file or directory

```
系统日志只有这两条
1、重启路由后，/var/etc/shadowsocksr.include 这个文件确实没有
2、/etc/init.d/shadowsockr start 后这个文件就出现了。
3、启动项目里 90 shadowsocksr

bigbighill, 2019-05-26:
新路由3 newifi3 d2

bigbighill, 2019-05-26:
没有办法，只好在启动项中 exit 0上方添加
/etc/init.d/shadowsocksr start

qyh214, 2019-06-15:
> 
> 
> 没有办法，只好在启动项中 exit 0上方添加
> /etc/init.d/shadowsocksr start

我尝试了重启，似乎无法复现，我这两天再测试下最新的，你尝试下清除所有配置后再试试。另外请提供下你使用的固件的提交hash

jliveusa, 2019-07-24:
同样遇到这个问题，要手动启动才可以开启ssrplus，用的官方openwrt

closed 2019-10-18 by coolsnowwolf

orange-cat-fantacy, 2020-03-17:
俺也一样

KimBin, 2020-04-01:
很简单 需要在“服务器” tab 里 启动服务器，就可以了。有gfw列表会自动选择的。

orange-cat-fantacy, 2020-04-02:
thinks qwq

rc452860, 2022-01-01:
修改`/etc/init.d/shadowsocksr`和`/usr/bin/ssr-rules`中替换`iptables `为`iptables -w `注意空格。

rc452860, 2022-01-01:
> 修改`/etc/init.d/shadowsocksr`和`/usr/bin/ssr-rules`中替换`iptables `为`iptables -w `注意空格。

该问题极有可能是因为iptables锁导致的。通过查询openwrt日志`loadread`，搜索`shadow`关键字
```
Sat Jan  1 20:43:01 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
Sat Jan  1 20:43:01 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
Sat Jan  1 20:43:01 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
Sat Jan  1 20:43:01 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
Sat Jan  1 20:43:01 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
Sat Jan  1 20:43:01 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
Sat Jan  1 20:43:02 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
Sat Jan  1 20:43:02 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
Sat Jan  1 20:43:02 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
Sat Jan  1 20:43:02 2022 user.err ssr-rules[2783]: Can't redirect, please check the iptables.
Sat Jan  1 20:43:02 2022 daemon.notice procd: /etc/rc.d/S95shadowsocksr: ssr-rules[2783]: Can't redirect, please check the iptables.
```
发现如下日志。

Jeremy-JM, 2022-04-19:
你是没理解到问题。问题说的是，已经配置好SSR，重启路由后，SSR未自动运行好。

whc23mj, 2022-04-21:
修改/etc/init.d/shadowsocksr中的START=99

Jeremy-JM, 2022-05-05:
> 
> 你是没理解到问题。问题说的是，已经配置好SSR，重启路由后，SSR未自动运行好。

问题出在，70版本的img，安装出来，系统空间太小了，如果使用FileBrowser或者可道云，又不修改默认配置，会导致上述问题和一些其他奇怪的问题。

zorrose82, 2022-08-17:
最近版本又出现了，设置了每周3-5点重启机器，结果每次ssrplus起不来，以前都要登陆web进去点一下ssrplus确认就好了
我直接在计划任务里加了一行，6点重启一下ssrplus
00 5 * * 3 sleep 5 && touch /etc/banner && reboot
00 6 * * 3  /etc/init.d/shadowsocksr  restart