coolsnowwolf/lede #1345: 编译极路由2完成后找不到固件

danbin, 2019-05-26:
极路由2 芯片：mt7620a， make menuconfig的时候型号已经选了hiWiFi HC5761了
编译完成后在bin/targets/ramips/mt7620/目录下找不到打包好的固件文件，请问是什么原因，有什么特别的配置吗？
编译输出如下，看起来编译结束没有错误：
Generating index for package ./transmission-remote-openssl_2.93-7_mipsel_24kc.ipk
Generating index for package ./transmission-web_2.93-7_mipsel_24kc.ipk
Generating index for package ./vpnbypass_1.3.1-1_all.ipk
Generating index for package ./wget_1.19.5-1_mipsel_24kc.ipk
Generating index for package ./wifischedule_1-2_all.ipk
Generating index for package ./zerotier_1.2.12-3_mipsel_24kc.ipk
Signing package index...
make[2]: Leaving directory '/home/hm/lede'
export MAKEFLAGS= ;make -w -r checksum
make[2]: Entering directory '/home/hm/lede'
make[2]: Leaving directory '/home/hm/lede'
make[1]: Leaving directory '/home/hm/lede'

danbin, 2019-05-26:
是这套源码不支持7620系列芯片吗？

asushugo, 2019-05-26:
是不是加东西太多，导致太大不生成固件？

danbin, 2019-05-26:
这个怎么确认是加东西太多的原因啊？我感觉加的不多

danbin, 2019-05-26:
也没报错啊

danbin, 2019-05-26:
群里说不支持mtk？但menuconfig里面有mtk的型号啊，openwrt原生也是支持mtk的啊。readme里面也没有给出支持列表啊

ope087, 2019-05-27:
往上找打包固件的日志啊,下面的这些日志都是正常的啊

closed 2019-10-18 by coolsnowwolf