coolsnowwolf/lede #1354: 建议添加KoolPorxy广告过滤

csrbzhb, 2019-05-27:
Koolproxy感觉是最好用的广告过虑软件，这个软件能过虑HTTPS的广告。建议添加这个工具支持。

lbbboy, 2019-05-27:
KP规则都没了。可以说已经GG了。

liwenjie119, 2019-05-27:
现在有个koolproxyR的项目，但是好像只对梅林固件，也没有大佬能整成通用插件

zxlhhyccc, 2019-05-27:
> KP规则都没了。可以说已经GG了。

适用kp（原作者一直在更新）
https://hosts.nfz.moe/full/hosts
http://dn-mwsl-hosts.qbox.me/hosts.txt
https://raw.githubusercontent.com/vokins/yhosts/master/hosts
https://easylist-downloads.adblockplus.org/easylistchina+easylist.txt
https://easylist-downloads.adblockplus.org/fanboy-social.txt
https://easylist-downloads.adblockplus.org/malwaredomains_full.txt
https://raw.githubusercontent.com/anudeepND/youtubeadsblacklist/master/domainlist.txt
https://fanboy.co.nz/r/fanboy-ultimate.txt
https://raw.githubusercontent.com/xinggs

lifetyper, 2019-05-28:
2.如果你自认为 Koolshare 论坛或者其固件的脑残粉，本人不欢迎你使用本源代码。所以如果你是，那么使用过程中遇到任何问题本人概不回应。

closed 2019-05-28 by coolsnowwolf

csrbzhb, 2019-05-28:
既然KoolProxy没有更新了，那哪个广告过虑软件最好用？求大神推荐。

Karmylr, 2021-09-03:
> 既然KoolProxy没有更新了，那哪个广告过虑软件最好用？求大神推荐。

说实话，都是垃圾