coolsnowwolf/lede #1357: ssr plus的编译选项在最新代码里不见了

WROIATE, 2019-05-28:
[image]
不太清楚为啥ssr plus的选项没了，有谁能说一下嘛？

signigelchan, 2019-05-28:
Love&Peace👇 .

WROIATE, 2019-05-28:
> Love&Peace👇 .

是直接没了？还是有芝麻开门？

signigelchan, 2019-05-28:
> > Love&Peace👇 .
> 
> 是直接没了？还是有芝麻开门？

我也是最近发现的。

WROIATE, 2019-05-28:
> > > Love&Peace👇 .
> > 
> > 
> > 是直接没了？还是有芝麻开门？
> 
> 我也是最近发现的。
lean不会是被喝茶了吧。。。

cuishitian77, 2019-05-28:
确实是没有了。不知道怎么回事，还是坐等答复吧

hanxi, 2019-05-29:
https://github.com/coolsnowwolf/lede/commit/b5daaadc0c6c17397efbc4d5c281c75b2bc571d7#commitcomment-33616316

cuishitian77, 2019-05-29:
谢了，找到了。

dongfangy2011, 2019-05-29:
> 谢了，找到了。

怎么找到了，我找不到

hanxi, 2019-05-29:
> > 谢了，找到了。
> 
> 怎么找到了，我找不到

看我发的链接啊

WROIATE, 2019-05-29:
> [b5daaad#commitcomment-33616316](https://github.com/coolsnowwolf/lede/commit/b5daaadc0c6c17397efbc4d5c281c75b2bc571d7#commitcomment-33616316)

感谢！

closed 2019-05-29 by WROIATE